gotthardp/lorawan-server #499: Multicast crash

amc2505, 2018-10-09:
Hello,

When I try to use multicast with loran 0.6.1
In my python program I use ...

ws.send(json.dumps({"devaddr":devaddr,"data":"01", "port":2, "time":"immediately"})).
devaddr, is defined in my program.

nothing happens...

when I look the logs ..

I have 

2018-10-09 13:48:57.289 [error] <0.784.0> Ranch listener http terminated with reason: badarg
2018-10-09 13:49:57.366 [debug] <0.800.0>@lorawan_connector_ws:websocket_init:95 WebSocket connector <<"To_Appserver">> with #{app => <<"Alarm-handler">>}
2018-10-09 13:49:57.381 [debug] <0.800.0>@lorawan_connector_ws:terminate:164 WebSocket terminated: {crash,error,badarg}
2018-10-09 13:49:57.384 [error] <0.800.0>@lorawan_handler:-mac_for_profile/1-fun-0-:318 CRASH REPORT Process <0.800.0> with 0 neighbours crashed with reason: bad argument in call to erlang:h$
2018-10-09 13:49:57.384 [error] <0.800.0> Ranch listener http terminated with reason: badarg

what could be the problem ?

thks.

gotthardp, 2018-10-09:
Hello. Could it be the case that one of the devices in the multicast group never sent an uplink?

amc2505, 2018-10-09:
yes .. it could be …it means that all the nodes in my group needs to have sent at least one uplink message ?

thanks.

gotthardp, 2018-10-09:
Right now yes, but I will make a fix for this in few hours.

gotthardp, 2018-10-09:
Fixed. Would you be able to build the server from sources and try the latest master, please?

gotthardp, 2018-10-09:
The class C (immediately) suffers from the same issue: if the device never sent an uplink, the server doesn't know via which gateway it is connected, so it cannot send class C or multicast.

gotthardp, 2018-11-16:
Fixed in 0.6.2

closed 2018-11-16 by gotthardp