gotthardp/lorawan-server #501: Buildroot question

amc2505, 2018-10-10:
Hello,

Is there an existing loran-server for build root ?

thanks.

gotthardp, 2018-10-10:
Hello. Not yet.

gotthardp, 2018-10-10:
But Erlang is there, so adding the lorawan-server *should not* be difficult. Is there a packet_forwarder for buildroot?

amc2505, 2018-10-10:
Not in build root, but I got one from another project (packet fwd) …I’m busy building Erlang but it seems to be OK … I had some problem with Openssl , but it seems to be solved…

Last but not least I need to be able to build lorawan-server...

amc2505, 2018-10-10:
> I will try to make the buildroot script for the lorawan-server

thanks… keep me informed…as soon as you have it I will test it...

amc2505, 2018-10-12:
If I have well understood, to include Lorawan-server in buildroot, I just need todo a mk,and a config.in  which copy some folders at the right place. right ? because Erlang is already there...

for the moment I did a buildroot image with erlang, packet-fwd  and python web socket-client and then I tar the loran-server on the working linux.. and this is OK.

Regards.

gotthardp, 2018-10-12:
Yes, Erlang is there, so you need to add the server only. I did some .mk and config.in, but ran out of disk space when testing it. :-( I need to empty some space first.

amc2505, 2018-10-17:
I can test it for you if you want ....

gotthardp, 2018-10-17:
I did put some draft scripts into "scripts/buildroot". It does compile and install into the target directory, but it is missing auto-start scripts. Would you be able to test it and contribute the rest, please? :)

amc2505, 2018-10-17:
yes, for sure …

Thks.

closed 2018-11-30 by gotthardp