gotthardp/lorawan-server #505: WebSocket Auth

JoobyPM, 2018-10-12:
Hi @gotthardp ,
As I understood there is no authentification for WebSocket connectors, is it true?
Could you please suggest – do you have in plans implementation of Basic Auth for WebSocket?

gotthardp, 2018-10-12:
Hello. The WS authentication is tricky and therefore is not implemented (yet).
As I understood e.g. [here](https://stackoverflow.com/questions/4361173/http-headers-in-websockets-client-api) the libraries support HTTP Basic only. The problem is the server doesn't store plaintext passwords (for security reasons), so the question is where to take the password from.

One solution would be to take the password from the connector configuration. The drawback would be that you could specify only one username-password per connection, and that it would be separated from the standard server users.

JoobyPM, 2018-10-12:
Thank you for the prompt reply.
It will be great if you can implement that feature.

gotthardp, 2018-10-12:
I just implemented something (without testing). Please give it a try and let me know if it doesn't work.

JoobyPM, 2018-10-12:
Many thanks for that enhancement.
I've builded and tested - it's working perfectly with HTTP Basic for WebSocket.

gotthardp, 2018-10-12:
Cool. Thanks for testing it.

closed 2018-10-12 by gotthardp