gotthardp/lorawan-server #507: at+recv event response taking too much time

gmbankar37, 2018-10-16:
Hi I am using RAK811 node and after sending a confirmed message I am getting a response after almost one minute even if the gateway is 10 meters away. Also I used to get a response in 4sec previously. What would be the reason for such behavior ?

gotthardp, 2018-10-16:
Hello. Could you please view the Frames list and filter only the affected device? How many uplinks and downlinks do you see?

gmbankar37, 2018-10-16:
[image]
[image]
[image]
[image]

So here are the frames who have different uplink timing regardless that I am sending them at a fixed time.

gotthardp, 2018-10-16:
But there are no downlinks, only uplinks. This explains why the confirmation takes so long. Could you please have a look into the debug.log to see if there are any errors or warnings?

gmbankar37, 2018-10-17:
Oh no I have send you only uplinks. There are also downlinks. here is the debug log for 2 days.

[debug.log.txt](https://github.com/gotthardp/lorawan-server/files/2485754/debug.log.txt)
[debug.log1.txt](https://github.com/gotthardp/lorawan-server/files/2485755/debug.log1.txt)

gotthardp, 2018-10-17:
And could you show me both uplinks and downlinks, please? For the same time, which is covered by the logs, e.g. 1 hour on 2018-10-16, between 18:00 and 19.

gmbankar37, 2018-10-17:
Apparantly I dont have data of it on the server saved in frames. Also during that time I had logs ignored as true.

gotthardp, 2018-10-17:
So how can we debug this issue?

gmbankar37, 2018-10-19:
I have two gateways with two different servers. I want to assign two devices for each but I dont want that two devices connected to a server to send any packet to another gateway. For now all the packets from 4 devices are received at 2 gateways. how can I disable a packet being received to particular gateway(not server) ?

gmbankar37, 2018-10-19:
[image]

My one node is getting connected in both RX1 and RX2. Why is this happening ?

gmbankar37, 2018-10-19:
[image]
What is this error? 
lorawan_mac:ingest_data_frame:92 <<"00000AB8">> Unexpected mtype 3
lorawan_mac:ingest_data_frame:92 <<"00000AB5">> Unexpected mtype 3
and these are the DevAddr which are not activated in the server?

gmbankar37, 2018-10-19:
> Please don't post many unrelated questions into one issue. It makes the
> discussion complicated.
>
>    1. You cannot filter frames sent by one gateway. All gateways
>    connected to a server are equally used.
>    2. The RX1 and RX2 alternation is an auto-selection feature. If RX1
>    fails, the server tries RX2.
>    3. "Unexpected mtype" means the device sent an unexpected mtype. For
>    example 3 = 011 binary is unconfirmed data down. Such frames should be sent
>    with other polarization (see the ipol parameter in packet_forwarder), so it
>    is not received by gateways.

First sorry for giving all issues in one thread.

1. I dont want to filter frames sent to one gateway I want to filter the no
of devices connected to one gateway. I want one node to be sending packet
to only one particular gateway.
2. But in the logs it says that it has accepted join request in RX1 why is
it going to RX2?

gotthardp, 2018-10-19:
1) This is currently not possible.
2) The server says it DID SEND a Join Accept, but the device then sends another Join Request, so the server thinks the device DIDN'T RECEIVE it and tries another RX window. In the profile you can fix either RX1 or RX2.