gotthardp/lorawan-server #509: Fcount question

amc2505, 2018-10-17:
hello,

Because I had strange behavior I have disable Fcnt on my profile for the moment. I have a Multicast which is in the same profile. It seems that the fcount are not deactivated for the multicast. If I deactivate fcount, it must be deactivated for the whole profile isn't it ?

gotthardp, 2018-10-17:
Hello. It's uplink FCnt **Check** that gets disabled, not the counter itself. The multicast is related to downlinks only. What exactly you want to disable?

amc2505, 2018-10-17:
In fact, in my testbed I have 7 modules which are listening in Multicast. i don’t know why, it was not working anymore, and there was nothing in the lorawan-server logs …I restarted the modules and nothing change… I put the multicast Fcount to zero and everything work.

What do you think about that ?

regards.

gotthardp, 2018-10-17:
The Multicast FCount in the server is permanent, but the stored FCount in the device is often not (although the standard says they should be). I suspect the Multicast may have been rejected by the device because the Fcnt had "wrong" value. Could it be a possible reason?

altishchenko, 2018-10-17:
It could. Met this a few times. Unless the device does something analogous
to 'reset on zero', communication breaks.

amc2505, 2018-10-17:
the fcount on the server side et device side must be the same or it must simply be bigger at server side ?

altishchenko, 2018-10-17:
I met devices that rejected the server's answers where FCntDown was way off. This can happen if you test the device on one server and then deploy it on another. The device in question should implement something analogous to "reset on zero" from the server side, otherwise without hard reset to the device you won't be able to connect it back - "security" prevails, thanks to LoRaWAN.

amc2505, 2018-10-19:
The problem is that, when there is a difference of about 16000, for loran multicast fcount and the device fcount this is not working anymore. 

So what happens …

I start the multicast system with some device … this is OK fcount increase and reach 16125 this is OK.

I had a new device listening on multicast ..because Lorawan fcount is more than 16000 and my device is 0 because it is new … it doesn’t work on the new device.

I reset the loran-server fcount to 



0, and then my new module work … but not the old one because again I have a difference of more than 16000.

So I think I need a system to reinit the fcount on a regular basis…..so I write a small program to re init the multicast fcount, but I got an http 405 error.

help is welcome..

gotthardp, 2018-10-19:
This is an interesting problem of the LoRaWAN standard. I am not sure how this should be solved; I will try asking the LoRaWAN Alliance. Let's see what the experts say.

amc2505, 2018-10-19:
and what about the fact that my python program is not able to reset multicast fcount, I ‘m able to change when I connect with the web interface.

gotthardp, 2018-10-19:
The manual reset to zero shall be possible. What HTTP request are you sending, what error are you getting and what is written in the log files?

amc2505, 2018-10-19:
I got ..

http_error 	{405,"/api/multicast_channels/ABBA0080/",{127,0,0,1}},

amc2505, 2018-10-19:
the address  "http://localhost:8080/api/multicast_channels/ABBA0080/« 

And I just try to upgrade doing a post 

    print(requests.post(addressServer,json=payload,auth=self.auth))

with payload

def set_payload(self):
        return [{"fcntdown":0}]

gotthardp, 2018-10-19:
1) You must use PUT, (not POST)
2) You must include the entire payload, i.e. do GET, change a value and then PUT everything back

amc2505, 2018-10-19:
ok…

Thanks.