gotthardp/lorawan-server #513: Connection with node get lost for several hours.

peticon, 2018-10-27:
I have build up two similar configuration with LoRa-Nodes (RM186 from Laird), RG1xx (Gateway from Laird) and different LoRaWAN-Server.
The LoRaWAN-Server are:
- your Server for local application
- TTN-Cloud-Server for Cloud applications
We would like to offer our customers both solutions:
Now we experience that with your Server the node-connection get sometimes lost for several hours.
With the TTN-Cloud-Server we don't observe such a behavior.
There are Errors reported in the log-file (collision-packet or too_early) and the node-status in the screenhot enclosed shows special incidences. 

My Question is:  Why does it take several hours in order to recover such incidences?
[image]
[image]

gotthardp, 2018-10-28:
Hello. The RX graph bellow doesn't show any gap. Does this mean the uplinks always work, but the downlink messages are not received by the node? How far is the device from the server? Did something change that could cause the drop in the downlink signal quality?

Concerning "collision-packet or too_early": how many gateways do you have?

peticon, 2018-10-30:
Hi, 
here the answers to your questions:
Does this mean the uplinks always work, but the downlink messages are not received by the node?
The uplinks get interrupted for several hours (see Device Status and Dashboard below). 
How far is the device from the server? 
The nodes at very close to the Gateway (about 8m).
Did something change that could cause the drop in the downlink signal quality?
No. Even on Weekend, when nobody is in the company, the contact get lost. 
Concerning "collision-packet or too_early": how many gateways do you have?
There is only one Gateway connected to the LWS-Server.

In the Events List there are a lot of entries "downlinks_lost". How can I disable downlinks?

[image]
[image]

gotthardp, 2018-10-30:
I dont see any gap on the dashboard. The collisions are because two devices did uplink at the same time. The gateway cannot send two downlinks hence the error.

peticon, 2018-10-30:
On the Dashboard you can see a join. This is the join for node 00001002. This node was calm (quiet) before for some hours. On the device Status you can see active and calm  times for the node 00001002.
How can I deactivate the Downloads?

peticon, 2018-11-09:
I mean downlinks not downloads in the last sentence of the entry before.
Do you see any chance to avoid transmission interrupts and so many "downlink losts" by modifying the Parameter Settings of the LoRaWAN-Server?
If yes, please suggest, which parameter should be changed. Please recommend setting values.
Thanks in advance!

gotthardp, 2018-11-09:
Try increasing Power and/or Set RX1 DR Offset in the device profile.