gotthardp/lorawan-server #514: Firewall Issues with port 8883

Tamburasca, 2018-10-27:
This is not a bug report, it is a wish: 

in order to circumvent FW issues with port 8883, would it also be possible to utilize MQTT with TLS client authentication on port 443 (ALPN), see also AWS's site on https://aws.amazon.com/de/blogs/iot/mqtt-with-tls-client-authentication-on-port-443-why-it-is-useful-and-how-it-works/

I'd be delighted on your reply. Thanx, Ralf A.

gotthardp, 2018-10-28:
Hello. You can switch to 443 in the configuration. The ALPN is also supported.

The request for MQTT is a duplication of https://github.com/gotthardp/lorawan-server/issues/459.

Tamburasca, 2018-10-30:
Hi Petr,

 

besides definining the environmental variables for the proxies in lorawan-server.service what needs to be done if the lorawan server can talk to the AWS IoT endpoint only over a proxy config? Thanks for your assistance.

 

Thanks, Ralf A.

 

Gesendet: Sonntag, 28. Oktober 2018 um 08:35 Uhr

gotthardp, 2018-10-30:
Forgive me my ignorance, please, but what proxies and proxy config do you mean? There is no such functionality in this server.

Tamburasca, 2018-10-30:
Hi Petr,

 

most likely did I not specify my problem in its entire detail.

 

The lorawan server is located on an EC2 instance in an AWS VPC with a firewall blocking it from the internet - being only openend to the corporate network. Hence, I cannot connect to the IoT, as there is also no endpoint service for IoT available by AWS.  However, we've got a proxy (port 80, 443, 8883) that can be used to get to the outside world. Can I utilize that proxy from the lorawan server such that I can connect to the AWS IoT endpoint? Thanks for your assistance and patience.

 

Best regards, Ralf A.
 

Gesendet: Dienstag, 30. Oktober 2018 um 08:15 Uhr

gotthardp, 2018-10-30:
Do you need to specify the proxy address or is it automatically intercepting the traffic? I assume you need to configure the proxy address somewhere, right?

gotthardp, 2018-10-30:
I see. And what protocol do you (want to) use when connecting to AWS IoT? HTTP or MQTT? Do you have to use MQTT, or could you use HTTP?
https://docs.aws.amazon.com/iot/latest/developerguide/protocols.html

Tamburasca, 2018-10-30:
Presently, our admins opened ports 80, 443, and 8883 with the proxy (Squid). To be on the safe side 80 and 443 are granted anyway. 8883 was opened on my request. 

 
 

Gesendet: Dienstag, 30. Oktober 2018 um 11:54 Uhr

https://docs.aws.amazon.com/iot/latest/developerguide/protocols.html

Tamburasca, 2018-10-30:
MQTT
 

Gesendet: Dienstag, 30. Oktober 2018 um 14:27 Uhr

But which protocol you want to use? HTTP or MQTT?

gotthardp, 2018-10-31:
I am afraid the library I am using does not support proxy.
https://stackoverflow.com/questions/46592826/is-it-possible-to-configure-emqttc-to-use-proxy-information