gotthardp/lorawan-server #520: Incorrect free memory display

xypron, 2018-11-17:
Before submitting a bug report please
 - read the Troubleshooting documentation;
 - make sure you are running the latest stable release;
 - inspect the debug.log to check if there are some warnings or errors reported.

xypron, 2018-11-17:
```
$ free
              total        used        free      shared  buff/cache   available
Mem:        2028220      119304      628256        8336     1280660     1807700
Swap:             0           0           0
```

http://192.168.1.99:8080/admin#/servers/list shows
free memory = 616 MB (31%)

I would have expected the available memory to be shown here because this is the relevant value to judge if the server is memory constricted.

Best regards

Heinrich

gotthardp, 2018-11-17:
But the "free" (628256 kB) is quite equal to 616 MB. Isn't it?

gotthardp, 2018-11-17:
The value displayed by the server is the total memory available to Erlang, whereas the `available` is the estimated memory available to new applications. These two values may differ.

xypron, 2018-11-17:
Linux uses unused memory for caching files to speed up operations. 628256 KiB is what is left over because not to many files have been read since the operating system was booted. The Linux operating system will reduce the size of the caches if any application wants more than another extra 628256 KiB.

So all applications together can use additional 1807700 KiB. This includes the running Erlang instance.

gotthardp, 2018-11-17:
The value displayed is provided by the Erlang system. There is no other value available.

xypron, 2018-11-17:
http://erlang.org/doc/man/memsup.html - On linux the memory available to the emulator is cached_memory and buffered_memory in addition to free_memory.

```
application:ensure_all_started(os_mon).
memsup:get_system_memory_data().
[{system_total_memory,2076897280},
 {free_swap,0},
 {total_swap,0},
 {cached_memory,1108701184},
 {buffered_memory,101879808},
 {free_memory,581124096},
 {total_memory,2076897280}]

memsup:get_memory_data().
{2076897280,1496158208,{<0.82.0>,320732}}

$ free -w -b
              total        used        free      shared     buffers       cache   available
Mem:     2076897280   151412736   580997120     8536064   101867520  1242619904  1821822976
Swap:             0           0           0
```

xypron, 2018-11-22:
This is the line to be corrected
priv/admin/admin.js:1283:
var free = 100 * entry['memory.free_memory'] / entry['memory.total_memory'];

gotthardp, 2018-11-22:
What should be used instead?

closed 2018-11-22 by gotthardp