gotthardp/lorawan-server #521: Cannot enter gateway location without Google API key

xypron, 2018-11-17:
Why should I want to share any location data with Google?
Please, provide text entry fields for the longitude and latitude as alternative to the entry via the map.

Best regards

Heinrich

xypron, 2018-11-25:
We could use OpenStreetMap instead of Google Maps. Then we can avoid having to buy a license key. A coding examples can be found in https://medium.com/@balramchavan/using-openstreetmap-inside-angular-v6-3d42cbf03e57

gotthardp, 2018-11-25:
The issue #454 also proposed that. Yes, we should get rid of google maps.

gotthardp, 2019-04-04:
I just switched the server to [Leaflet](http://angular-ui.github.io/ui-leaflet/#!/), which by default uses the OpenStreetMap. The Google API key is no longer needed.

closed 2019-04-04 by gotthardp