gotthardp/lorawan-server #530: uplinks_missed problem

jithinsisaac, 2018-11-22:
Hey.. I am forwarding packets from my gateway to another machine on LAN which has the server installed. I get a uplinks_missed error. This inspite of the fact that i am receiving the frames form my end node, am able to see the payload. what could be the reason?
[image]

matiasz8, 2018-11-22:
What is the content of your Parse Uplink?

gotthardp, 2018-11-22:
Hey. You receive some, but miss other uplinks. Your gateway reports uplinks-lost. Is there something moe detailed in the GATEWAY log?

jithinsisaac, 2018-11-23:
Can you let me know how to check Gateway log in the command line? 
Also, as shown in the attached pic, 
[image]
why isn't the gateway showing IP address and last alive status?
This is a Single channel gateway.. Is that the reason?

Because, i have another setup with a multi channel gateway, and the entire setup with the server works fine.

sn3kyJ3di, 2022-04-10:
What did you end up doing with this?  I just bought and setup an Advantech WISE-6610 with a handful of sensors and I get a lot of uplinks_missed warnings in my event console.  I believe these are coming from a Node-to-Server FCnt mismatch but I'm not sure.