gotthardp/lorawan-server #531: Downlink

dencorpz1, 2018-11-23:
Hello
How can I set the frequency and sf for downlink package?
Sometimes you need to send a package to the device at a different frequency or SF than the standard settings.
Is it possible to add the ability to specify these parameters in the downlink package?
like so
{"data": "00", "port": 2, "time": "immediately", "frequency": 868.9, "SF": 12}
You also need the ability to specify the mac address of the base station from which to send the packet.
These parameters are needed at the stage of debugging equipment.
When the profile changes, the RX2 parameters do not change immediately.

gotthardp, 2018-11-23:
The frequency and SF is controlled by the server based on the LoraWAN specification. Feel free to implement any proprietary extensions you need.

dencorpz1, 2018-11-23:
Can an example how to do this?

dencorpz1, 2018-11-27:
why?