gotthardp/lorawan-server #537: Class A and Class C behaviour

gmbankar37, 2018-11-27:
I have implemented Class A and Class C devices. But I dont see any difference in the implementation of both. With reduced Latency in Class C should it be helpful to avoid collision? 
Can you please explain the following condition: 
Class A and Class C are sending a frame every one sec and get an acknowledgement from gateway after 3-4 secs. Does the module have a buffer to store uplink frame while waiting for the acknowledge to come in. How will these two devices behave in such situation?

gotthardp, 2018-11-27:
Class A devices listen for downlink just after an uplink. Class C devices do exactly the same, but *in addition* to that they also listen constantly for "immediate" downlinks. If your Class C devices sends uplink and then awaits ACK, then it behaves like a Class A device-- the implementation is identical.

gmbankar37, 2018-11-27:
So when and how are "immediate" downlinks generated?

gotthardp, 2018-11-27:
Via a Backend Connector (see the documetation).

closed 2019-03-16 by gotthardp