gotthardp/lorawan-server #540: Https connector: cannot specify root CA

xypron, 2018-11-29:
When trying to connect to my HTTPS server I receive the errors below. It is unclear to me where to put the root CA with which the server certificate was signed.
```
2018-11-29 11:16:04.008 [debug] <0.335.0>@lorawan_connector_http:ensure_gun:159 Connecting TTT to https://myserver:4433
2018-11-29 11:16:04.238 [info] <0.340.0> TLS client: In state hello received SERVER ALERT: Fatal - Handshake Failure
2018-11-29 11:16:09.261 [info] <0.343.0> TLS client: In state hello received SERVER ALERT: Fatal - Handshake Failure
2018-11-29 11:16:14.338 [info] <0.346.0> TLS client: In state hello received SERVER ALERT: Fatal - Handshake Failure
2018-11-29 11:16:19.364 [info] <0.349.0> TLS client: In state hello received SERVER ALERT: Fatal - Handshake Failure
2018-11-29 11:16:24.450 [info] <0.352.0> TLS client: In state hello received SERVER ALERT: Fatal - Handshake Failure
2018-11-29 11:16:29.490 [info] <0.355.0> TLS client: In state hello received SERVER ALERT: Fatal - Handshake Failure
2018-11-29 11:16:29.490 [warning] <0.335.0>@lorawan_connector_http:handle_info:125 Connector TTT failed: {shutdown,{tls_alert,"handshake failure"}}
```

xypron, 2018-11-29:
I digged in further its is not the CA posing the problem. But OTP 21 disabled RSA key exchange.

xypron, 2018-11-29:
ssl:connect("server",4433,[{ciphers, ssl:cipher_suites(all)}]) works fine but it seems that ther is no possibility to add cipher_suites in the lorawan-server yet.

altishchenko, 2018-11-29:
there is an ssl_options field for the sys.config. it defaults to [] in .app.src. you can put that there.

xypron, 2018-11-29:
@altishchenko 

Thanks Alexander for the hint. Unfortunately in http://erlang.org/doc/man/ssl_app.html, chapter "Configuration" there is no configuration parameter relating to ciphers mentioned.

Regards

Heinrich

altishchenko, 2018-11-30:
Hello Heinrich (@xypron),
Options in this field are used in connection setup (in lorawan_connector_http.erl) and really correspond to options defined in this page: http://erlang.org/doc/man/ssl.html - ssl_option() data type.

altishchenko, 2018-11-30:
And yes, this field is for the lorawan_server application environment, not for ssl app. You can see it referenced in .app.src file in the server's src/ directory, so you can add it along  {websocket_timeout, 3600000} in the sys.config.

altishchenko, 2018-11-30:
@xypron and just as thought (may be @gotthardp Petr will agree to it too) - at this list is defaulted to the transport used, but may be it will be better to give all ciphers instead? Like:
lorawan_connector_http.erl:285: Opts = application:get_env(lorawan_server, ssl_options, ssl:cipher_suites(all)),
???

xypron, 2018-11-30:
@altishchenko 
Thanks for all the clarification. I think doc/Connectors.md should be updated to indicate that the config parameter ssl exists and is passed in lorawan_connector_http to gun.

Always allowing weak ciphers is a bad idea. Such weak ciphers could be used in a man in the middle attack so using ssl:cipher_suites(all) is a bad idea.

The same parameter ssl is also used for slack messages in send_slack_raw(). This is also undocumented.

ssl options may include information about the certificate used to sign into a https connection. Therefore it would make sense if the ssl options could be specified per backend connector. Please, add this to the wishlist.

Best regards

Heinrich