gotthardp/lorawan-server #543: join accept could not received

arigus, 2018-12-03:
Tested older version could get join accept and with new version of server device could not get JOIN ACCEPT, i see join marker in the dashboard it is repeating. Gateway status warning dwell_time_voilated.  i check timing is 5s in windows1 and  6s  in window2. 
Please,advice.

arigus, 2018-12-03:
[image]

arigus, 2018-12-03:
device added to Activated(nodes) list as expected. but device do not receive JOIN ACCEPT.

arigus, 2018-12-03:
here is debug log:
2018-12-03 17:37:18.732 [info] <0.4449.0> device 0000670DD1BEBAB9 {join,<<"A237DB4B">>}
2018-12-03 17:37:18.732 [warning] <0.4449.0> node A237DB4B {repeated_reset,80}
2018-12-03 17:37:18.732 [debug] <0.4449.0>@lorawan_mac:encode_accept:409 Join-Accept <<"A237DB4B">>, netid <<18,52,81>>, cflist [], rx1droff 0, rx2dr 0, appkey <<"C54907BA992295DA425CD3D1152B9C87">>, appnce <<"020884">>
2018-12-03 17:37:18.733 [debug] <0.4449.0>@lorawan_handler:join:117 Join-Accept in RX1: {rxq,868.5,<<"SF7BW125">>,<<"4/5">>,{{2018,12,3},{9,32,40.898426}},undefined,1895301971,-73,6.5}
2018-12-03 17:37:25.710 [info] <0.4454.0> device 0000670DD1BEBAB9 {join,<<"A237DB4B">>}
2018-12-03 17:37:25.710 [warning] <0.4454.0> node A237DB4B {repeated_reset,81}
2018-12-03 17:37:25.711 [debug] <0.4454.0>@lorawan_mac:encode_accept:409 Join-Accept <<"A237DB4B">>, netid <<18,52,81>>, cflist [], rx1droff 0, rx2dr 0, appkey <<"C54907BA992295DA425CD3D1152B9C87">>, appnce <<"71E6B0">>
2018-12-03 17:37:25.711 [debug] <0.4454.0>@lorawan_handler:join:120 Join-Accept in RX2: {rxq,868.3,<<"SF7BW125">>,<<"4/5">>,{{2018,12,3},{9,32,47.952429}},undefined,1902353483,-77,7.5} {0,0,868}
2018-12-03 17:37:32.946 [info] <0.4456.0> device 0000670DD1BEBAB9 {join,<<"A237DB4B">>}
2018-12-03 17:37:32.946 [warning] <0.4456.0> node A237DB4B {repeated_reset,82}
2018-12-03 17:37:32.946 [debug] <0.4456.0>@lorawan_mac:encode_accept:409 Join-Accept <<"A237DB4B">>, netid <<18,52,81>>, cflist [], rx1droff 0, rx2dr 0, appkey <<"C54907BA992295DA425CD3D1152B9C87">>, appnce <<"3FFE8B">>
2018-12-03 17:37:32.946 [debug] <0.4456.0>@lorawan_handler:join:117 Join-Accept in RX1: {rxq,868.1,<<"SF7BW125">>,<<"4/5">>,{{2018,12,3},{9,32,55.021262}},undefined,1909404995,-75,8.3}

gotthardp, 2018-12-03:
The log seems to be generated by some weird sever version, Which version are you using?

arigus, 2018-12-03:
lorawan_server-0.6.0.

gotthardp, 2018-12-03:
Did you compile it yourself? The log mentions "lorawan_mac:encode_accept:409", but there is no log command on line 409 in any of the versions I released. Could your version be modified or broken somehow?

arigus, 2018-12-04:
yes, compiled it myself, but without modify any. and just download new version and compiled. JOIN ACCPET is not fixed. Log was as bellow:  
2018-12-04 00:05:02.772 [debug] <0.811.0>@lorawan_mac:encode_accept:441 Join-Accept <<"A3072790">>, netid <<18,52,81>>, cflist [], rx1droff 0, rx2dr 0, appkey <<"C54907BA992295DA425CD3D1152B9C87">>, appnce <<"C0E870">>
2018-12-04 00:05:02.772 [debug] <0.811.0>@lorawan_handler:join:117 Join-Accept in RX2: {rxq,868.5,<<"SF7BW125">>,<<"4/5">>,{{2018,12,3},{16,0,24.449734}},undefined,2103903787,-71,7.8} {0,0,868}
2018-12-04 00:05:09.826 [info] <0.814.0> device 0000670DD1BEBAB9 {join,<<"A3072790">>}

gotthardp, 2018-12-04:
What is your "RX2 Freq" configuration in the Infrastructure/Network and Device/Profile? I can see you entered "868". As I suggested above, please re-review your configuration. The standard defaults (you have to enter) are documented in https://github.com/gotthardp/lorawan-server/blob/master/doc/Regions.md