gotthardp/lorawan-server #544: GatewayID as only numbers

ToniStonde, 2018-12-04:
If gatewayID (MAC) is only numbers LoRaWAN gives unknown mac error and displays the address as '0000000000000000' even when MAC is correctly inserted for the gateway.

gotthardp, 2018-12-04:
Could you send me a screenshot of the MAC being numbers only, please?

ToniStonde, 2018-12-04:
[image]

In the gateway itself:

option gateway_ID '3338333041005200'

If I replace any number in that with 1 letter all OK. So im not sure its the cause the lora_pkt_forwarder itself or LoRaWAN but atleast in the device side all seems to be OK.

gotthardp, 2018-12-04:
That's weird, but possible. Would you please be able to capture the JSON structure sent by the gateway to the server? Either in the gateway's log (if it's logged) or by the Wireshark tool.

ToniStonde, 2018-12-04:
Well Wireshark only shows the json like this:

{"txpk":{"powe":16,"imme":false,"tmst":2166238616,"codr":"4/5","datr":"SF12BW125","freq":869.525,"modu":"LORA","rfch":0,"ipol":true,"size":13,"data":"YOZfKxIBBgAGcy/RXw=="}}

Do you have your own gateway which you could test in your own env? So just change the GatewayID to only numbers and then restart the service which forwards the packets?

ToniStonde, 2018-12-04:
Well actually i think the bug is in the lora_pkt_fwd:

"gateway_ID":"333833304b005200"

"gateway_ID":3.3383330410052e+15

After the lora_pkt_fwd parses the configs if only numbers it displays it like that in the bottom. If it has atleast one letter, then OK (like the upper example)

gotthardp, 2018-12-04:
so, it's a lora_pkt_fwd bug, shall I close this issue?

ToniStonde, 2018-12-04:
I think so. Sorry for bothering :)

gotthardp, 2018-12-04:
That's OK :)

closed 2018-12-04 by gotthardp