gotthardp/lorawan-server #548: More than one multicast group for one device

JoobyPM, 2018-12-05:
Right now, I found only one way to add one device to the multiple multicast groups, by creating the profile for each device.

Maybe there is another way to make it?

JoobyPM, 2018-12-05:
[image]

[image]

An end-device MAY support up to 4 multicast groups «Multicast Channels». 
For example:
> the device `A` should be in associated with following multicast channels - `G1`,`G2`,`G3`,`G4`
> and
> the device `B` should be in associated with following multicast channels - `G2`,`G8`,`G9`.
>> Devices `A` and `B` are identical, if not that case, might use just one profile.
To do so, I have to create an individual profile for devices `A` and `B`.

Our application dynamically creates groups, according to user «device groups», to implement that feature we have to create duplicates of profiles to be able to assign devices to different channels.

gotthardp, 2018-12-05:
How it's on the firmware level? The multicast numbers are hardcoded, right? You build new firmware for each device? Or you have less hardcoded numbers?

JoobyPM, 2018-12-06:
Depends on implementation. Within our implementation (fork of the Lmic), we
can have up to 4 multicast addresses (with mutual sessions keys) and 1
broadcast channel (with deducated session keys).
All this configurations can be changed on application level.

gotthardp, 2018-12-06:
I see, that makes sense.