gotthardp/lorawan-server #554: Support of the FUOTA LoRaWAN specs

JoobyPM, 2018-12-08:
Since September LoRaWAN published specs:
- LoRaWAN Application Layer Clock Synchronization Specification v1.0.0
- LoRaWAN Fragmented Data Block Transport Specification v1.0.0
- LoRaWAN Remote Multicast Setup Specification v1.0.0

Some of the commercial server going to support it next beginning of the year (might be some of the themes already supports). It going to be great if lorawan-server supports that specs.

gotthardp, 2018-12-08:
That's a lot of work. Anyone else out there that needs some of these features? Please let me know.

WillooWisp, 2020-06-30:
Fragmented data block transport for FUOTA would be a nice feature, that we will need very soon. Any plans on adding this, similar to what chirpstack has done?

WillooWisp, 2020-08-31:
Any news regarding any thoughts about implementing these standards?

yehessa, 2023-05-21:
Hello @gotthardp 

> That's a lot of work. Anyone else out there that needs some of these features? Please let me know.

Hello @gotthardp 
Thank you for your Job. I'm a new user.
Of course I will appreciate this function.

Thank you for your help

Best