gotthardp/lorawan-server #558: FR: NewChannelReq

altishchenko, 2018-12-11:
Hello Petr!
Any plans on implementing NewChannelReq MAC command? I found that it is used by Actility/ThingPark on ABP device's initial connect or reset on zero - they use it to setup additional channels much like CFList on join does. I found it pretty helpful instead of hardcoding all possible channels into the device itself.

Consider this track of the initial message exchange, few more messages, not shown here, follow, setting up more channels after it:

(Mind, this is RU864 band, so just 2 base channels not 3).

Actility -> Node:
===========
Mtype: UnconfirmedDataDown
RX1/RX2Delay: 2000
Flags: ADR : 0, ADRAckReq : 0, ACK : 1, FPending : 0
Mac (hex): 03520300010523389d840702e8d983500703b8e18350

MAC.Command.LinkADRReq
MAC.LinkADRReq.DataRate_TXPower : 0x52
MAC.LinkADRReq.DataRate_TXPower.DataRate : 5
MAC.LinkADRReq.DataRate_TXPower.TXPower : 2
MAC.LinkADRReq.ChMask : 0x0003
MAC.LinkADRReq.Redundancy : 0x01
MAC.LinkADRReq.Redundancy.ChMaskCntl : 0
MAC.LinkADRReq.Redundancy.NbTrans : 1
MAC.Command.RXParamSetupReq
MAC.RXParamSetupReq.DLsettings : 0x23
MAC.RXParamSetupReq.DLsettings.RX1DRoffset : 2
MAC.RXParamSetupReq.DLsettings.RX2DataRate : 3
MAC.RXParamSetupReq.Frequency (Hz) : 869100000
MAC.Command.NewChannelReq
MAC.NewChannelReq.ChIndex : 2
MAC.NewChannelReq.Frequency (Hz) : 864100000
MAC.NewChannelReq.DrRange : 0x50
MAC.NewChannelReq.DrRange.MaxDR : 5
MAC.NewChannelReq.DrRange.MinDR : 0
MAC.Command.NewChannelReq
MAC.NewChannelReq.ChIndex : 3
MAC.NewChannelReq.Frequency (Hz) : 864300000
MAC.NewChannelReq.DrRange : 0x50
MAC.NewChannelReq.DrRange.MaxDR : 5
MAC.NewChannelReq.DrRange.MinDR : 0

Node -> Actility:
===========
Mtype: ConfirmedDataUp
Flags: ADR : 1, ADRAckReq : 0, ACK : 0
Mac (hex): 0307050707030703

MAC.Command.LinkADRAns
MAC.LinkADRAns.Status : 0x07
MAC.LinkADRAns.Status.ChannelMaskAck : 1
MAC.LinkADRAns.Status.DataRateAck : 1
MAC.LinkADRAns.Status.PowerAck : 1
MAC.Command.RXParamSetupAns
MAC.RXParamSetupAns.Status : 0x07
MAC.RXParamSetupAns.Status.ChannelACK : 1
MAC.RXParamSetupAns.Status.RX2DataRateACK : 1
MAC.RXParamSetupAns.Status.RX1DRoffsetACK : 1
MAC.Command.NewChannelAns
MAC.NewChannelAns.Status : 0x03
MAC.NewChannelAns.Status.DataRateRangeOK : 1
MAC.NewChannelAns.Status.ChannelFrequencyOK : 1
MAC.Command.NewChannelAns
MAC.NewChannelAns.Status : 0x03
MAC.NewChannelAns.Status.DataRateRangeOK : 1
MAC.NewChannelAns.Status.ChannelFrequencyOK : 1

gotthardp, 2018-12-12:
Hello. It depends on the interest from other users. For now I didn't need this command.

altishchenko, 2018-12-13:
Ok. I find it really useful for ABP devices ADR control. ABPs don't have the CFList option and sometimes have only basic channels pre-programmed in their code. So, if you intend to use extra channels you either need to go OTAA or use this command. I've started my experiment on it, but I am having trouble locating the proper place in the code where this can be inserted.

altishchenko, 2018-12-13:
@gotthardp is build_fopts a good place to start with NewChannelReq? For me  it looks like the place, can you confirm?

gotthardp, 2018-12-13:
It depends how the user should interact with it, but `build_fopts` is where the MAC commands are generated.

altishchenko, 2018-12-13:
Well, we have a channel list in the network ADR setup, given there is a way to detect that ABP device has just restarted (reset on zero, for ex,), network server may choose to set the extra channels of the device. It will not possibly happen during one reply and can possibly be few replies with more and more channels set, and one final command to change the channel mask to match. Like they do in actility - set mask to default channels, add new channels, set mask to all channels in the next downlink. It will require some more state to be kept. Not sure if it will affect the database too.

ZuratonE, 2019-02-15:
Please, add this command.
It could take channels from network channel list, that didnt fit into the CFlist field in JoinACK.
Now we have end nodes with 2 basic channels, and 5 channels for CFlist. And we have 16 channel base station, and cannot use other 9 free channels without hardcoding them into nodes.

mitwave, 2019-07-23:
Please, add this command!!
I want to use 16ch GW on AS923 (Japan).

I tried to add ch on "Networks" -> "Edit network #XXXX" -> "Channels" tab.
However, if I add over than 5ch (6~14(=16-default ch (2ch))), server does not send CFlist to device... (Device channel list is empty except default ch...)

if I can use 16ch GW on latest version, would you give me the instruction how to setup?

stefanobimbo, 2019-09-02:
Please, add this command!!

I use ABP devices with ADR disabled. Each device is pre-configured for using extra channels and not only standard ones (I use Telit RE866A1-EU module for LoRaWAN stack) but what it seems is that devices don't use these channels if not requested by the network server. I cannot Join OTA, and cannot use ADR (customer requirements), so NewChannelFreq is the only way (as far as I know).