gotthardp/lorawan-server #560: GNSS and lorawan server

jot4p, 2018-12-11:
Hi 

It's possible to implement in the future GNSS on your loraserver ?
 
http://www.multitech.net/developer/software/mlinux/using-mlinux/gpsgnss-on-conduit/

It's a very interesting to use lora like tracker of small devices. 

Thanks.

altishchenko, 2018-12-12:
@jot4p I am sorry, but what it has to do with the server? It is a feature of the Multitech packet forwarder that is not relayed over to the server. LoRa was initially designed to be **untrackable** during the WW2. You can track only if you use the FSK modulation (to some extent).

jot4p, 2018-12-12:
Hi @altishchenko , 

Please check this: https://www.loraserver.io/loraserver/features/geolocation/

Thanks.

gotthardp, 2018-12-12:
@jot4p, you mean Geolocation, right? Geolocation will be supported once there is a low cost gateway with precise timestamping. So far, all gateways that support geolocation were terribly expensive. I know Semtech did some change to their IP policy, which may lead to creation of low cost gateways with geolocation support. Once this happen, I will add geolocation to the server.

jot4p, 2018-12-13:
Hi @gotthardp , 

I have the multiple gateways with gnss support (multitech).  

All new multitech gateways support this.