gotthardp/lorawan-server #571: Channel configuration

pmst, 2018-12-21:
I am trying to enable 8 channel on the NS but I am not able to input the frequency when doing the add:

- if I type in 867.1 I get an error like "Please type a number"
- if I type in 867,1 I get "Error: Frequency must be 400..1000 MHz"

I searched in the documentation but no solution found so far.

altishchenko, 2018-12-21:
Hmm, works ok with commas and dots for me. What version of the server you are using? Did you try refreshing browser cache?

pmst, 2018-12-21:
I am currently on 0.6.4, I tried with different browser and cleared up the cache...same result...maybe I am doing something else wrong...

pmst, 2018-12-21:
I tried also on a new VM setting different locale values, same behavior.

altishchenko, 2018-12-21:
I mean, it is your browser/system locale that may have effect. Of your local computer.

gotthardp, 2018-12-27:
Hello. Same here. I am not able to reproduce this error, even when I set 'comma' as a decimal separator. Java appears to be ignoring Window settings and keeps using the decimal dot. As said above, please indicate what is your OS, browser and what decimal separator is configured in your system.

pmst, 2019-01-14:
Hello, tried with both the following systems:
- OS: Window7, Browser Firefox and Chrome, decimal separator ,
- OS: Debian, Browser Iceweasel, decimal separator . (locale en_US)

I always get the error that frequency must be 400 MHz...10000MHz

pmst, 2019-01-14:
Ok I found out what was going on...in the ADR tab for some reason the "Initial RX2 Freq (MHz) *" was wrong (not sure why...) and this was causing the error (it was that frequency wrong not the channel one). The fact that the error was the same tricked me into the wrong direction.

closed 2019-01-14 by pmst