gotthardp/lorawan-server #573: BUG: connector MQTT localhost json_syntax_error

xypron, 2019-01-07:
The following command leads to input that is accepted as JSON downlink
```
mosquitto_pub -h localhost -t dragino/downlink -m '{"deveui" : "1234567812345678", "data" : "123"}'
```

```
mosquitto_pub -h localhost -t dragino/downlink -m '{"deveui" : "1234567812345678", "data" : "aGVsbG8="}'
```
leads to  **connector MQTT localhost json_syntax_error**

According to https://jsonlint.com/ both messages are legal JSON.

xypron, 2019-01-07:
The problem is **not** in the JSX library:

```
10> jsx:decode( << "{\"deveui\" : \"1234567812345678\", \"data\" : \"123\"}" >>, [return_maps, {labels, atom}]).
#{data => <<"123">>,deveui => <<"1234567812345678">>}
11> jsx:decode( << "{\"deveui\" : \"1234567812345678\", \"data\" : \"aGVsbG8=\"}" >>, [return_maps, {labels, atom}]).
#{data => <<"aGVsbG8=">>,deveui => <<"1234567812345678">>}
```

gotthardp, 2019-01-07:
The `data` should be a hexa-string.

xypron, 2019-01-07:
Yes. the bug is caused by *lorawan_admin:parse()*.

If we have a downlink function we should not make any fancy assumptions about the format of a field called data.