gotthardp/lorawan-server #577: delayed gateway network delay graph

pdl86, 2019-01-08:
Hello 

in the gateway status page,  the network ddelay graph is "delayed" by almost 1.5 day .
[image]

the above screen shot was taken today : Jan 08 at 5:55PM , and the latest network delay displayed is dated back jan 7th 08:32 AM
the  linux machine running the lorawan server has the correct date.

Any idea ? 

regards
Philippe

pdl86, 2019-01-08:
Hello 

without doing nothing, "suddenly" the Network delay graph catch up somewhat but strangely : 
[image]

pdl86, 2019-01-08:
Note that the Network delay at 60/100 msec is "normal" as this gateway is linked to the server via a low bw/high latency link., but the display of all  delays vertically is not really normal.

pdl86, 2019-01-08:
It seems that the first issue is related to the fact that the "remote" gateway with the  packet forwarder does not have "joined" devices, and the second graph pops up when a device join the lorawan sevrer ( via the gateway) .

gotthardp, 2019-01-29:
The delay is calculated upon downlink. If there is no downlink, the delay cannot be calculated. The graph was not delayed, it simply didn't move because there were no downlinks.

closed 2019-03-16 by gotthardp