gotthardp/lorawan-server #578: Add basic auth for API

hallard, 2019-01-08:
Peter,
Would it be possible to add basic auth (I'm using SSL) to the REST API
I need to request data from a browser with JQuery (or any js) and Digest Auth lib, but none of them work. I tried and implemented :

 - `https://github.com/inorganik/digest-auth-request`
 - `https://github.com/Kynec/digest-ajax`

Curl with digest is working fine, so the problem is not on the server side but on client side (JS) and also, a lot of people have trouble with this javascript implementation.

Since this kind of problem is also mentioned in #502 and #564 so adding option to switch to basic auth would save some lot of time, including me.
Thanks

gotthardp, 2019-01-08:
Hello. Yes, I will re-introduce Basic Auth pretty soon. I need it myself.

hallard, 2019-02-12:
Waiting this to be implemented, is there any way to disable authentication? 
I know it's not secure but It's for testing on local net only.
thanks

gotthardp, 2019-02-12:
Are you able to git pull and test the latest HEAD, please? It's easier to implement it than to disable it :)

hallard, 2019-02-13:
Yes I'm able but the final target is a multitech conduit so need to recompile whole image. Is it done in 0.6.5 version ?

gotthardp, 2019-02-13:
Yes, it is in the 0.6.5 I released few hours ago.

hallard, 2019-02-13:
Awesome, thanks, how do we enable it ?

gotthardp, 2019-02-13:
No configuration. Just send the Basic Authentication header in your request.

closed 2019-02-15 by gotthardp