gotthardp/lorawan-server #581: Packet Forwarder Bridge

jot4p, 2019-01-15:
Hi, 

This topic is not an issue, but I need help to know if it's possible or not this:

I'm using  "@gotthardp lorawan server" for multiple devices, but now I have 3 devices provided by intelilight (flashnet.ro) to control light on the street. 
This devices use lorawan TYPE C protocol and requires that my multitech packet forwarder dedicated for intelilight server: "serv_port_down”: 50122, “serv_port_up”: 50122,“server_address”: “fc1.intelilight.eu”

This is impossible for me because my gateways are sending the data to gotthardp server.

It's possible to use the server to redirect the data "simulating packet forwarder" to the intelilight server ? Send and Receive Downlinks and provide this data for the nodes ?

Thanks

altishchenko, 2019-01-15:
Server will not act as a forwarder as such, but it is quite possible to find **data protocol** specification for your devices (types of messages they send and receive) and implement appropriate handlers/connectors at your own server. Normally vendors will publish this data or some good sole on the internet already hacked the protocol. Strangely, I couldn't get anything from the flashnet.ro site you mentioned, seems to be down...

jot4p, 2019-01-15:
The Idea is receive and send the data without change anything.. yeah.. i'm trying..

jot4p, 2019-01-22:
Hi, 

Does anyone recommend a solution to this problem?

gotthardp, 2019-01-29:
The "intelilight server" is a LoRaWAN server or an application server? I am not sure I understand the issue.

jot4p, 2019-02-04:
This is not an issue properly....

InteliLight is based on this:
https://intelilight.eu/wp-content/uploads/2015/06/InteliLIGHT_LoRaWAN-architecture-.png

based on the webpage:

> LoRaWAN™ compatible network server
> The LoRaWAN™ compatible network server manages and prioritizes all the network elements, radio >and information flow. Every inteliLIGHT® controller must be registered with a local network operator in >order to be able to connect to the central management software. The inteliLIGHT® LoRaWAN™ >compatible controller is already integrated with most LoRaWAN™ public operators

https://intelilight.eu/technology/intelilight-lorawan-compatible-street-lighting-remote-management/

But I don't have any EUI/APPKey etc... This works only if I use packet forwarder directly for the intelilight server:

		"serv_port_down": 50122,
		"serv_port_up": 50122,
		"server_address": "fc1.intelilight.eu",

Do you have any idea how I can integrate this with your server ?

gotthardp, 2019-02-04:
In my server you'd need to setup a backend connector for the InteliLight server. The LoRaWAN standard 1.0.x does not specify inter-server interface, so being LoRa compliant doesn't help with the InteliLight connectivity. I think the InteliLight datasheet might been too creative about interoperability of their solution ;-)

closed 2019-03-16 by gotthardp