gotthardp/lorawan-server #583: Can't send downlink using HTTP

smalrad, 2019-01-17:
Hello,
I can't send downlinks using HTTP. All I obtain is a 404 answer.
The doc says to send the POST to /in/{devaddr}: doesn't work
I saw an existing closed issue with a POST made to /down/{devEUI}: doesn't work

Probably a configuration issue but I don't find the reason.
Below the curl request done:

`[root@localhost lorawan-server]# curl -v --digest -uadmin:admin -H "Content-Type: application/json" -X POST http://192.168.1.1:8080/in/70B3D58FF0033ABF -d "{\"data\":\"04000100000000000000\", \"port\":2, \"time\":\"immediately\"}"
* About to connect() to 192.168.1.47 port 8080 (#0)
*   Trying 192.168.1.47...
* Connected to 192.168.1.47 (192.168.1.47) port 8080 (#0)
* Server auth using Digest with user 'admin'
> POST /in/70B3D58FF0033ABF HTTP/1.1
> User-Agent: curl/7.29.0
> Host: 192.168.1.47:8080
> Accept: */*
> Content-Type: application/json
> Content-Length: 0
>
< HTTP/1.1 404 Not Found
< content-length: 0
< date: Thu, 17 Jan 2019 09:07:35 GMT
< server: Cowboy
<
* Connection #0 to host 192.168.1.47 left intact
```

`

altishchenko, 2019-01-17:
@smalrad Did you create an incoming HTTP connector as documented in doc/Connectors.md?
You should have a URI field with just `http:` in it and Received Topic set to `/in/{devaddr}`

altishchenko, 2019-01-18:
Oh, and also I noticed multiple double quotes. You should encapsulate your message in **single** quotes using shell. Like this: `-d '{"data":"04000100000000000000", "port":2, "time":"immediately"}'`

smalrad, 2019-01-18:
@altishchenko I have conformation from the device manufacturer that the port to use is 2. I also used singles quotes as recommended (in fact the double quotes pb was in the copy/paste of the code. Internal " were escaped with \.
Still doesn't work. I also tried with a Murata eval board from STM, and I have the same issue.
The packet seems correctly received by the packet_forwarder:
JSON down: {"txpk":{"powe":0,"imme":true,"codr":"4/5","datr":"SF9BW125","freq":868.1,"modu":"LORA","rfch":0,"ipol":true,"size":23,"data":"YLCgTVeAKAACg7z9YAfpsoxrvP9L+AM="}}
INFO: [down] a packet will be sent in "immediate" mode
INFO: tx_start_delay=1493 (1493.562500) - (1497, bw_delay=1.500000, notch_delay=1.937500)

Could that be an issue in ADR configuration, or Channels in the "network" ?

smalrad, 2019-01-21:
> @altishchenko I have confirmation from the device manufacturer that the port to use is 2. I also used singles quotes as recommended (in fact the double quotes pb was in the copy/paste of the code. Internal " were escaped with backslashes.
> I am also confused by the same problem,have you solved the question :"Can't send downlink using HTTP" ,
> my test environment is CN470.
> I can send downlink text by websocket ,but can't see message from gateway .as follwing .
> If you have any good suggestions ,please tell me .many thanks.

To make sure, I have registered the device to Objenious, a class C capable French public network: downlink messages works well. Thus I believe it is either a bug or configuration issue on lorawan-server

workholic365, 2019-01-22:
@smalrad thanks,I will try and check configurations again.

workholic365, 2019-01-22:
@smalrad I have tried again,it works.thanks.

smalrad, 2019-01-24:
I did the test with a public LoraWan network (Objenious in France), and class C message works well.
I can only point out an issue on LoraWan-server.
Any debug log to look at  ?

gotthardp, 2019-01-29:
@smalrad if you still have this issue, please post here the configuration of your connector

smalrad, 2019-01-30:
Hi Petr,

Yes I still have the issue for sending class C downlink.

 

Below the screen shot of my [ultra basic] « in » connector :

 

 

Would be nice if you can help.

 

Best regards

Stephan

altishchenko, 2019-01-31:
@smalrad Sorry, can't see your attachments... Does you Received Topic start with the '/'? debug.log file will be helpful also.

smalrad, 2019-01-31:
Hi Alexander,

Yes the topic is « /in/{devaddr} ».

Moreover this connector works well for class A downlink. 

Sending a class C downlink I see the packet reaching the packet forwarder and transmitted overthe air, but apparently not understood by the device.

 

I will make new tests and collect debug.log if you wish.

 

Best regards

Stephan

gotthardp, 2019-02-01:
If the Class C is actually transmitted, but not "seen" by the device, there can be a bug in the device. Are you able to debug the device to see if/why it was rejected by the device?

gotthardp, 2019-02-04:
Well. The Class C must be sent in the RX2 window, which uses different transmission parameters. That is correct. Please check if your **device configuration** of RX2 the same as the server config. That might be the problem. In which LoRa region you are?

gotthardp, 2019-12-25:
This one appears to be solved.

closed 2019-12-25 by gotthardp