gotthardp/lorawan-server #584: uplink fail (re-up)

BouTayehGaby, 2019-01-22:
hello!

Why my gateway is failing to upload data to the app server (red arrow, when I hover the mouse over it, it says "re-up")? I checked the configuration of my handler and my connector it seems correct. I tried to use postman to simulate and uplink to my app server, and the app server is working just fine.
Another unrelated question, what is the field "confirmed"?
[image]

Thank you

gotthardp, 2019-01-29:
Hello. Re-Up means "repeated uplink". All your uplink frames have FCnt=0, so the server considers them as a retransmissions of one uplink frame.
"Confirmed" means "Confirmed uplink", i.e. request for ACK (see LoRaWAN standard for more details.)

BouTayehGaby, 2019-01-30:
In fact, they all have Fcnt 0 since In my code I am sending only one frame and then the sensor stops, there is no loop. (just for testing). You can see in the figure below that these frames are different (they have different payloads)
[image]

What could possibly cause a fail in uplink? The other Application (Pysens Sensor, seen in the figure) is able to transmit the data to the app server without any problem, however, the GPSsensorHandler fails to do so, although the data is being sent to the same web application.

BouTayehGaby, 2019-01-30:
I put the code in a loop and tried to send multiple frames successively, it Worked! Am I missing something?  Why would it work for multiple transmissions but not for a single one?

altishchenko, 2019-01-31:
@BouTayehGaby Hello, your gateway has nothing to do with the FCnt counter, that is your device that is sending the data uplinks and not increasing the Up counter, hence the server ignores the data as ducplicate. You can check if that is the correct data by disabling the FCnt check in your device's Profile.

gotthardp, 2019-02-01:
Yeah. The server ignores retransmitted frames, so if the FCnt is the same it does not sent anything to backend. Try setting the "FCnt check" in your device profile to "Disabled".

BouTayehGaby, 2019-02-01:
Thank you. Problem Solved, you may close this issue.

closed 2019-02-01 by gotthardp