gotthardp/lorawan-server #585: MQTT multicast problem

amc2505, 2019-01-22:
hello,

I try to initiate a downlink message using MQTT. This is working with devices. But when I try to send a multicast message using the multicast device, this doesn't work, I got a message in the log ... invalid device or something like this ...
Is it by design or is it a bug ?
thanks.
Regards.

gotthardp, 2019-01-29:
Hello. That should work, so it's an error if it doesn't.

gotthardp, 2019-01-30:
The code for both cases is identical. It may be due to a misconfiguration. What is your connector configuration?

amc2505, 2019-01-30:
> Répondre à : gotthardp/lorawan-server <reply@reply.github.com>

MQTT
When I choose a device which is in my device list this is OK.

When I choose the device which is created in my multicast list it says that the device is unknown.

[cid:image001.png@01D4B88C.BE041DA0]

amc2505, 2019-01-30:
Is the same configuration, publish uplink is ok.

Publish event is not ok.

De : Alain-michel CIETERS <am.cieters@noolitic.com>
Date : mercredi 30 janvier 2019 à 11:12
À : gotthardp/lorawan-server <reply@reply.github.com>
Objet : Re: [gotthardp/lorawan-server] MQTT multicast problem (#585)

MQTT
When I choose a device which is in my device list this is OK.

When I choose the device which is created in my multicast list it says that the device is unknown.

[cid:image001.png@01D4B88C.BE041DA0]