gotthardp/lorawan-server #586: No events to my mqtt client

stefanobimbo, 2019-01-24:
I use 0.6.4
I configured an MQTT connector as follows:

[image]

I can see uplink messages in my mqtt subscriber  and send downlink message from my mqtt publisher. Downlink messages arrive to my end node, so I'm shure both uplink and downlink topics work OK.
The problem is that I don't receive events on "ClasseC/events" topic.
I aspect to receive event messages when my confirmed dowlink message in either delivered or lost.

amc2505, 2019-01-25:
I  have the same issue.. uplink is ok with mqtt but not events

gotthardp, 2019-01-29:
That is weird. It shall work as you said. Any warning/error in the server log?

amc2505, 2019-01-29:
> Répondre à : gotthardp/lorawan-server <reply@reply.github.com>

Hello,

Nothings in the logs….

Thks.

gotthardp, 2019-02-01:
Oh, that is indeed a bug. I made a fix. Would you be able to build the server from sources and test it, please? Or you want to wait for a release?

amc2505, 2019-02-05:
I can wait for the next release. but i'm not alone.

thks.

regards

moneybag, 2019-02-12:
same problem here, using mosquitto

gotthardp, 2019-02-12:
@moneybag, and can you test the latest MASTER, please? it should be fixed there

gotthardp, 2019-02-13:
Fixed in v0.6.5

closed 2019-02-13 by gotthardp

amc2505, 2019-02-13:
Hi,

When will you build a new release ?

Thks.

Date : mercredi 13 février 2019 à 10:47

Closed #586<https://github.com/gotthardp/lorawan-server/issues/586>.

gotthardp, 2019-02-13:
I just did.

amc2505, 2019-02-13:
Thks.

Date : mercredi 13 février 2019 à 10:58

romansoft, 2019-03-02:
I'm still seeing this issue in 0.6.7. Has it not been pushed yet?