gotthardp/lorawan-server #587: Empty data replay to a confirmed downlink

stefanobimbo, 2019-01-24:
I use 0.6.4 configured with an mqtt connector.

I send a confirmed downlink message (devaddr is C0000001):
{"port":1, "time":"immediately", "confirmed":true, "data":"2010", "receipt":"1"}

After the message arrives to my end node and it is ack'ed, an uplink message is sent to my mqtt subscriber with an empty data field:
{"codr":"4/5","data":"","datetime":"2019-01-24T09:19:54Z","datr":"SF9BW125","devaddr":"C0000001","fcnt":4,"freq":868.1,"lsnr":11.0,"rssi":-49}

Is this correct?
How can I use the replay message to know which downlink message was ack'ed (I use a class C end node, so I can queue more than one downlink messages at a time, and then waiting for the ack for each of them) 
See also issue #586

gotthardp, 2019-01-29:
It is correct to receive an empty uplink that was sent to ACK a downlink. However, you should have also received an **event** message with the receipt ID. Did you? If not, then it's Issue #586.

gotthardp, 2019-03-16:
Feel free to reopen if the issue persists.