gotthardp/lorawan-server #588: re-creating the database

bashhole, 2019-01-25:
Hello Petr I faced a problem, I use docker-compose and everything that in the folder "storage" I mount in the host server when I do docker-compose down and UP always creates new base how to force not to create base, and to check whether there is in the folder current and to use it

services:
  t_lw:
    image: t_lw:latest
    volumes:
      - t_data_lw:/storage/
    ports:
      - 1680:1680
      - 8080:8080
    restart: always
volumes:
    t_data_lw: {}

root@test:/var/lib/docker/volumes/telemetric_t_data_lw/_data# ls -la
total 16
drwxr-xr-x 4 root root 4096 Jan 25 15:44 .
drwxr-xr-x 3 root root 4096 Jan 25 14:04 ..
drwxr-xr-x 2 root root 4096 Jan 25 16:08 Mnesia.lorawan@232085a29835
drwxr-xr-x 2 root root 4096 Jan 25 15:43 Mnesia.lorawan@4c60030ab0c7

gotthardp, 2019-01-28:
Hello. The directory name is based on a machine/container ID (232085a29835 or 4c60030ab0c7). If you create container with the same name, the same Mnesia directory will be re-used when existing. Did you set both name and hostname parameters as described here: https://github.com/gotthardp/lorawan-server/blob/master/doc/Docker.md?

bashhole, 2019-01-29:
Thank you very mutch Petr, i can try this method)

gotthardp, 2019-01-29:
OK. Please let me know (and close this issue) once the issue is solved.

bashhole, 2019-01-30:
Thanks Petr i add in docker-compose.yml my service 'hostname: lorawan' Worked database not recreated

closed 2019-01-30 by bashhole