gotthardp/lorawan-server #591: Received Frames Order

hallard, 2019-01-31:
Hi Petr,

we noticed sometime the frame received display order looks wrong. Take a look at the picture below, looks like the console display the ACK downlink before the uplink (the FCnt looks to confirm that). Is there any option to also display the ms in the time field so we can check to be sure ?

[image]

Thanks

altishchenko, 2019-01-31:
@hallard Hello Charles, this is not a protocol issue, this is due to the asynchronous nature of the internal event processing (frame received, ack sent) in the code. There is nothing wrong in the communication there :)

hallard, 2019-01-31:
@altishchenko,
yes of course I agree, comms are okay, but the displayed order is driving me mad debuging issue 

may be I missed something you can help me on(I scheduled a downlink on a device)
on the picture below  

- 11:31:03, the first confirmed uplink come
- 11:31:04, ACK is send 1 second later on the receive slot, makes sense
- 11:32:12, my downlink is sent (but I don't see any uplink)
- 11:32:12, uplink is coming same time of the downlink
- 11:32:43, new uplink classic send
- 11:32:44, the confirmed uplink come on 1s receive window

[image]

gotthardp, 2019-02-01:
Hello. It is sorted based on the "Time" and the Time has only "seconds", so frames sent in the same second are missorted. One would need to add milliseconds to the timestamp to fix this.

altishchenko, 2019-02-05:
@hallard Yeah, this should drive you mad %) Unfortunately, the only way to properly decipher this is to take FCnt into account... Then you will see, that your downlink (logged **first**) is actually a response to your uplink (logged **next**)... Some manufacturers promise milli/micro/nano second precision on the frame timestamps in the nearest future. Not sure if it will help though.

gotthardp, 2019-02-13:
I made a fix for this. You may want to try the MASTER branch, or wait for a next release.

closed 2019-02-13 by gotthardp