gotthardp/lorawan-server #592: dbexport different port

bashhole, 2019-02-05:
Hello how do I export if I have a different port?

root@a9270261552:/usr/lib/lorawan-server/bin# ./dbexport 
http://localhost:8080/api/users:
2019-02-05 12:54:30 ОШИБКА 404: Not Found.
http://localhost:8080/api/gateways:
2019-02-05 12:54:30 ОШИБКА 404: Not Found.
http://localhost:8080/api/networks:
2019-02-05 12:54:30 ОШИБКА 404: Not Found.
http://localhost:8080/api/multicast_channels:
2019-02-05 12:54:30 ОШИБКА 404: Not Found.
http://localhost:8080/api/profiles:
2019-02-05 12:54:30 ОШИБКА 404: Not Found.
http://localhost:8080/api/devices:
2019-02-05 12:54:30 ERROR 404: Not Found.
http://localhost:8080/api/nodes:
2019-02-05 12:54:30 ERROR 404: Not Found.
http://localhost:8080/api/ignored_nodes:
2019-02-05 12:54:30 ERROR 404: Not Found.
http://localhost:8080/api/handlers:
2019-02-05 12:54:30 ERROR 404: Not Found.
http://localhost:8080/api/connectors:
2019-02-05 12:54:30 ERROR 404: Not Found.

gotthardp, 2019-02-06:
Which server version are are you running?
Could you please post few last lines from the server log showing the error message?

bashhole, 2019-02-06:
Version 0.5.5 
/var/log/lorawan-server/error.log Nothing not say from dxexport

bashhole, 2019-02-06:
./dbexport creates a backup folder and creates json files inside, but they are empty

gotthardp, 2019-02-07:
Is there *any* error in the log file? There must be at lest the 404 Not Found error. If there is *no* error, then there is some other server listening on the port 8080.

bashhole, 2019-02-07:
i get http://185.127.25.13:8080 or http://localhost:8080
output from web
" Cannot GET /"

gotthardp, 2019-02-10:
What does `netstat -tlp` print for port 8080. I am pretty sure there is something else listening, or the server installation is broken.

bashhole, 2019-02-18:
I figured it out myself, just went to the page and copied the json

closed 2019-02-18 by bashhole