gotthardp/lorawan-server #593: Sending downlink to many class C devices at the same time

stefanobimbo, 2019-02-08:
I use 0.6.4
I want to send downlink messages to many class C nodes (hundreds) at the same time. 
I am testing with only two nodes and sometimes I see the following log:

2019-02-08 10:48:36.925 [debug] <0.24324.7>@lorawan_application_backend:try_class_c:277 C0000001 delaying Class C downlink

What does it mean? Does lorawan-server delay downlink due to gateway not available? Does lorawan-server calculates also gateway duty cycle when checking its availability?

gotthardp, 2019-02-10:
The message means that a class A downlink was expected (after an uplink), but you sent a class C downlink to that device, so it had to wait for the RX2 window to open.
If you want to send class C to many devices I suggest you use multicast.

stefanobimbo, 2019-02-11:
I use Telit RE866A1-EU module as end node. This module doesn't support multicast, so I have to send separate downlink messages to each node using class C capability of immediate sending. Can there be problems queuing handreds of downlink request (one for each node)? Can gateway discard sending due to its duty cycle? Have you any hint to suggest?

gotthardp, 2019-02-11:
First of all, please note that there are **legal** regulations that require every device (including a gateway!) to transmit for 0.1 or 1% of time. It is not fair to steal all bandwidth-- there may be other devices on this frequency who want to send something.
There is no class C queueing in the gateway (packet_forwarder) nor in my server. You will need to send the "immediate" downlink one by one, keeping enough time between the transmissions to not violate the regulations in your country.

closed 2019-03-16 by gotthardp