gotthardp/lorawan-server #594: Gateway reconnects every several minutes

romansoft, 2019-02-09:
I'm seeing issue similar to "Infinite gateway reconnect #377". First, one gateway started constantly reconnecting every few minutes. Now there are two. The other gateways are still working normally.
I don't see any issues with gateway backhaul connectivity and latency is low <50ms.

```
2019-02-08 21:34:55.234 [info] <0.333.0> gateway 00800000A00037B7 {connected,{{216,168,52,14},25730}}
2019-02-08 21:37:13.261 [info] <0.333.0> gateway 647FDAFFFE0057A6 {connected,{{216,168,52,14},13793}}
2019-02-08 21:40:25.663 [info] <0.333.0> gateway 00800000A00037B7 {connected,{{216,168,52,14},4683}}
2019-02-08 21:40:53.522 [info] <0.333.0> gateway 647FDAFFFE0057A6 {connected,{{216,168,52,14},33708}}
2019-02-08 21:44:43.952 [info] <0.333.0> gateway 647FDAFFFE0057A6 {connected,{{216,168,52,14},10944}}
2019-02-08 21:45:56.084 [info] <0.333.0> gateway 00800000A00037B7 {connected,{{216,168,52,14},64484}}
2019-02-08 21:48:34.192 [info] <0.333.0> gateway 647FDAFFFE0057A6 {connected,{{216,168,52,14},58105}}
2019-02-08 21:51:26.303 [info] <0.333.0> gateway 00800000A00037B7 {connected,{{216,168,52,14},11042}}
2019-02-08 21:52:24.682 [info] <0.333.0> gateway 647FDAFFFE0057A6 {connected,{{216,168,52,14},4018}}
```

Using version 0.6.4.

romansoft, 2019-03-02:
Any idea why this might be happening?

gotthardp, 2019-03-16:
Not really. It means the gateway did change its port number. This could be e.g. due to a NAT between the server and the gateway.