gotthardp/lorawan-server #596: how to start the server lorawan?

schwiderke92, 2019-02-14:
I would like to use lora-server with my gateway, but I can not do it. I'm still a new rookie.
I think of creating tutorial on youtube about it, to help the community, could someone help me?

gotthardp, 2019-02-14:
Video tutorial would be cool. So far we have only written documentation. The README.md is a good place to start: https://github.com/gotthardp/lorawan-server/blob/master/README.md#usage

schwiderke92, 2019-02-14:
I have managed to start the server, but I can not find the server  adress  to put on the  my gateway

gotthardp, 2019-02-14:
Retrieve the IP address of your machine using `ifconfig`

schwiderke92, 2019-02-14:
can it conflict, if I run the server by virtual machine ?, I got the ip and setei in the gateway, but it does not connect

gotthardp, 2019-02-14:
There can be a firewall, for example. Try ping or ssh first to check if routing works and make sure the firewall is correctly setup or disabled.

closed 2019-02-26 by gotthardp