gotthardp/lorawan-server #597: Strange issue with 0.6.5 and MQTT Connector

hallard, 2019-02-15:
We've built the latest 0.6.5 version on multitech as usual and it's works fine.

Then we imported with dbimport and since that web interface is crashing (no more listening on port 8080) even if all the rest looks like to work

I investigated an found it's elated on connectors side, so I deleted database, imported back all except connectors.json and then created connectors by hands. It works fine with the 2 following connectors:

[image]

But as soon as I create the 3rd connector as follow, the server crash, that's strange because it using the same handler than the websocket one.

[image]

here are the log of the server 
```
2019-02-15 16:09:40.264 [error] <0.491.0> device 0CCDB79A4167656E unknown_deveui
2019-02-15 16:10:42.000 [error] <0.535.0> device 000397A16165656E unknown_deveui
2019-02-15 16:11:10.268 [error] <0.539.0> device 0001B79A4164665E unknown_deveui
2019-02-15 16:11:31.761 [error] <0.543.0>@lorawan_connector:node_to_vars:15 gen_server <0.543.0> terminated with reason: no function clause matching lorawan_connector:node_to_vars({node,<<2,11,39,119>>,<<"gabriel_profile">>,<<"Bleu">>,<<64,82,246,107,80,66,197,145,120,167,91,...>>,...}) line 15
2019-02-15 16:11:31.940 [error] <0.543.0>@lorawan_connector:node_to_vars:15 CRASH REPORT Process <0.543.0> with 0 neighbours crashed with reason: no function clause matching lorawan_connector:node_to_vars({node,<<2,11,39,119>>,<<"gabriel_profile">>,<<"Bleu">>,<<64,82,246,107,80,66,197,145,120,167,91,...>>,...}) line 15
2019-02-15 16:11:31.972 [error] <0.396.0>@lorawan_connector:node_to_vars:15 Supervisor lorawan_connector_sup had child {mqtt,<<"gabriel-MQTT">>} started with lorawan_connector_mqtt:start_link({connector,<<"gabriel-MQTT">>,<<"gabriel_handler">>,<<"json">>,<<"mqtt://127.0.0.1:1883">>,undefined,...}) at <0.543.0> exit with reason no function clause matching lorawan_connector:node_to_vars({node,<<2,11,39,119>>,<<"gabriel_profile">>,<<"Bleu">>,<<64,82,246,107,80,66,197,145,120,167,91,...>>,...}) line 15 in context child_terminated
2019-02-15 16:11:32.052 [error] <0.548.0>@lorawan_connector:node_to_vars:15 gen_server <0.548.0> terminated with reason: no function clause matching lorawan_connector:node_to_vars({node,<<2,11,39,119>>,<<"gabriel_profile">>,<<"Bleu">>,<<64,82,246,107,80,66,197,145,120,167,91,...>>,...}) line 15
2019-02-15 16:11:32.085 [error] <0.548.0>@lorawan_connector:node_to_vars:15 CRASH REPORT Process <0.548.0> with 0 neighbours crashed with reason: no function clause matching lorawan_connector:node_to_vars({node,<<2,11,39,119>>,<<"gabriel_profile">>,<<"Bleu">>,<<64,82,246,107,80,66,197,145,120,167,91,...>>,...}) line 15
2019-02-15 16:11:32.121 [error] <0.396.0>@lorawan_connector:node_to_vars:15 Supervisor lorawan_connector_sup had child {mqtt,<<"gabriel-MQTT">>} started with lorawan_connector_mqtt:start_link({connector,<<"gabriel-MQTT">>,<<"gabriel_handler">>,<<"json">>,<<"mqtt://127.0.0.1:1883">>,undefined,...}) at <0.548.0> exit with reason no function clause matching lorawan_connector:node_to_vars({node,<<2,11,39,119>>,<<"gabriel_profile">>,<<"Bleu">>,<<64,82,246,107,80,66,197,145,120,167,91,...>>,...}) line 15 in context child_terminated
2019-02-15 16:11:32.197 [error] <0.549.0>@lorawan_connector:node_to_vars:15 gen_server <0.549.0> terminated with reason: no function clause matching lorawan_connector:node_to_vars({node,<<2,11,39,119>>,<<"gabriel_profile">>,<<"Bleu">>,<<64,82,246,107,80,66,197,145,120,167,91,...>>,...}) line 15
2019-02-15 16:11:32.230 [error] <0.549.0>@lorawan_connector:node_to_vars:15 CRASH REPORT Process <0.549.0> with 0 neighbours crashed with reason: no function clause matching lorawan_connector:node_to_vars({node,<<2,11,39,119>>,<<"gabriel_profile">>,<<"Bleu">>,<<64,82,246,107,80,66,197,145,120,167,91,...>>,...}) line 15
2019-02-15 16:11:32.269 [error] <0.396.0>@lorawan_connector:node_to_vars:15 Supervisor lorawan_connector_sup had child {mqtt,<<"gabriel-MQTT">>} started with lorawan_connector_mqtt:start_link({connector,<<"gabriel-MQTT">>,<<"gabriel_handler">>,<<"json">>,<<"mqtt://127.0.0.1:1883">>,undefined,...}) at <0.549.0> exit with reason no function clause matching lorawan_connector:node_to_vars({node,<<2,11,39,119>>,<<"gabriel_profile">>,<<"Bleu">>,<<64,82,246,107,80,66,197,145,120,167,91,...>>,...}) line 15 in context child_terminated
2019-02-15 16:11:32.291 [error] <0.396.0> Supervisor lorawan_connector_sup had child {mqtt,<<"gabriel-MQTT">>} started with lorawan_connector_mqtt:start_link({connector,<<"gabriel-MQTT">>,<<"gabriel_handler">>,<<"json">>,<<"mqtt://127.0.0.1:1883">>,undefined,...}) at <0.549.0> exit with reason reached_max_restart_intensity in context shutdown
2019-02-15 16:11:32.302 [error] <0.395.0> Supervisor {<0.395.0>,lorawan_backend_sup} had child connectors started with lorawan_connector_sup:start_link() at <0.396.0> exit with reason shutdown in context child_terminated
2019-02-15 16:11:32.609 [error] <0.553.0>@lorawan_connector:node_to_vars:15 gen_server <0.553.0> terminated with reason: no function clause matching lorawan_connector:node_to_vars({node,<<2,11,39,119>>,<<"gabriel_profile">>,<<"Bleu">>,<<64,82,246,107,80,66,197,145,120,167,91,...>>,...}) line 15
2019-02-15 16:11:32.633 [error] <0.553.0>@lorawan_connector:node_to_vars:15 CRASH REPORT Process <0.553.0> with 0 neighbours crashed with reason: no function clause matching lorawan_connector:node_to_vars({node,<<2,11,39,119>>,<<"gabriel_profile">>,<<"Bleu">>,<<64,82,246,107,80,66,197,145,120,167,91,...>>,...}) line 15
2019-02-15 16:11:32.676 [error] <0.550.0>@lorawan_connector:node_to_vars:15 Supervisor lorawan_connector_sup had child {mqtt,<<"gabriel-MQTT">>} started with lorawan_connector_mqtt:start_link({connector,<<"gabriel-MQTT">>,<<"gabriel_handler">>,<<"json">>,<<"mqtt://127.0.0.1:1883">>,undefined,...}) at <0.553.0> exit with reason no function clause matching lorawan_connector:node_to_vars({node,<<2,11,39,119>>,<<"gabriel_profile">>,<<"Bleu">>,<<64,82,246,107,80,66,197,145,120,167,91,...>>,...}) line 15 in context child_terminated
2019-02-15 16:11:32.753 [error] <0.554.0>@lorawan_connector:node_to_vars:15 gen_server <0.554.0> terminated with reason: no function clause matching lorawan_connector:node_to_vars({node,<<2,11,39,119>>,<<"gabriel_profile">>,<<"Bleu">>,<<64,82,246,107,80,66,197,145,120,167,91,...>>,...}) line 15
2019-02-15 16:11:32.774 [error] <0.554.0>@lorawan_connector:node_to_vars:15 CRASH REPORT Process <0.554.0> with 0 neighbours crashed with reason: no function clause matching lorawan_connector:node_to_vars({node,<<2,11,39,119>>,<<"gabriel_profile">>,<<"Bleu">>,<<64,82,246,107,80,66,197,145,120,167,91,...>>,...}) line 15
2019-02-15 16:11:32.814 [error] <0.550.0>@lorawan_connector:node_to_vars:15 Supervisor lorawan_connector_sup had child {mqtt,<<"gabriel-MQTT">>} started with lorawan_connector_mqtt:start_link({connector,<<"gabriel-MQTT">>,<<"gabriel_handler">>,<<"json">>,<<"mqtt://127.0.0.1:1883">>,undefined,...}) at <0.554.0> exit with reason no function clause matching lorawan_connector:node_to_vars({node,<<2,11,39,119>>,<<"gabriel_profile">>,<<"Bleu">>,<<64,82,246,107,80,66,197,145,120,167,91,...>>,...}) line 15 in context child_terminated
2019-02-15 16:11:32.870 [error] <0.555.0>@lorawan_connector:node_to_vars:15 gen_server <0.555.0> terminated with reason: no function clause matching lorawan_connector:node_to_vars({node,<<2,11,39,119>>,<<"gabriel_profile">>,<<"Bleu">>,<<64,82,246,107,80,66,197,145,120,167,91,...>>,...}) line 15
2019-02-15 16:11:32.893 [error] <0.555.0>@lorawan_connector:node_to_vars:15 CRASH REPORT Process <0.555.0> with 0 neighbours crashed with reason: no function clause matching lorawan_connector:node_to_vars({node,<<2,11,39,119>>,<<"gabriel_profile">>,<<"Bleu">>,<<64,82,246,107,80,66,197,145,120,167,91,...>>,...}) line 15
2019-02-15 16:11:32.929 [error] <0.550.0>@lorawan_connector:node_to_vars:15 Supervisor lorawan_connector_sup had child {mqtt,<<"gabriel-MQTT">>} started with lorawan_connector_mqtt:start_link({connector,<<"gabriel-MQTT">>,<<"gabriel_handler">>,<<"json">>,<<"mqtt://127.0.0.1:1883">>,undefined,...}) at <0.555.0> exit with reason no function clause matching lorawan_connector:node_to_vars({node,<<2,11,39,119>>,<<"gabriel_profile">>,<<"Bleu">>,<<64,82,246,107,80,66,197,145,120,167,91,...>>,...}) line 15 in context child_terminated
[... cut ...]
2019-02-15 16:11:37.781 [error] <0.581.0> Supervisor {<0.581.0>,lorawan_backend_sup} had child connectors started with lorawan_connector_sup:start_link() at <0.594.0> exit with reason reached_max_restart_intensity in context shutdown
2019-02-15 16:11:37.788 [error] <0.387.0> Supervisor {<0.387.0>,lorawan_sup} had child backends started with lorawan_backend_sup:start_link() at <0.581.0> exit with reason shutdown in context child_terminated
2019-02-15 16:11:37.795 [error] <0.387.0> Supervisor {<0.387.0>,lorawan_sup} had child backends started with lorawan_backend_sup:start_link() at <0.581.0> exit with reason reached_max_restart_intensity in context shutdown
```
Any hints on what's going on ?

gotthardp, 2019-02-15:
Hello. This is a bug. Could you please post the crash.log as well? It contains more details. Also, what did happen before the crash (uplink, downlink, test button press)?

hallard, 2019-02-15:
It's happening when I change the handler in the connectors list, I tried to create new based on old one and it crashed even. Looks like searching something in config that does not find or match. It's not happening on receiving, just when trying to change configuration. Then when restarting the server, the server start but not listening on 8080 and not responding anymore. I tried a lot except restarting from scratch (no backup restore)

Here the last crash log

````
2019-02-15 17:44:24 =ERROR REPORT====
** Generic server <0.352.0> terminating
** Last message in was nodes_changed
** When Server state == {state,{connector,<<"gabriel-MQTT">>,<<"gabriel_handler_new">>,<<"json">>,<<"mqtt://127.0.0.1:1883">>,undefined,<<"gabriel/up">>,<<"gabriel/events">>,undefined,<<>>,<<>>,true,undefined,undefined,undefined,undefined,undefined,undefined,undefined,[],0,0,undefined},[{<<"mqtt://127.0.0.1:1883">>,[]},undefined,undefined,undefined],{<<>>,[]},{<<"gabriel/up">>,[]},{<<"gabriel/events">>,[]},{<<>>,[]},[]}
** Reason for termination ==
** {function_clause,[{lorawan_connector,node_to_vars,[{node,<<2,11,39,119>>,<<"gabriel_profile_new">>,<<"Bleu">>,<<64,82,246,107,80,66,197,145,120,167,91,56,63,94,36,130>>,<<84,137,32,197,119,148,142,187,183,127,243,140,228,137,62,181>>,<<"Mike">>,undefined,3,4,{{2019,2,15},{14,51,54}},{{2019,2,15},{14,51,54}},0,{{2019,2,15},{14,52,45}},[{<<0,128,0,0,160,0,18,168>>,{rxq,867.7,<<"SF12BW125">>,<<"4/5">>,{{2019,2,15},{14,51,54.402131}},1234277533401,2641425236,-45,7}}],1,{undefined,undefined,undefined},{0,0,[{0,7}]},[],undefined,{0,0,869.525},[],[{-45,7},{-67,6.8},{-44,5.8},{-42,6.5}],undefined,undefined,undefined,[],undefined,undefined,undefined,undefined}],[{file,"/home/transnumeric/mlinux-4.0.1/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/lorawan-server/0.6.5-r0/git/src/lorawan_connector.erl"},{line,15}]},{lorawan_connector_mqtt,'-build_hierarchy/3-fun-0-',4,[{file,"/home/transnumeric/mlinux-4.0.1/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/lorawan-server/0.6.5-r0/git/src/lorawan_connector_mqtt.erl"},{line,52}]},{lists,foldl,3,[{file,"lists.erl"},{line,1263}]},{lorawan_connector_mqtt,handle_info,2,[{file,"/home/transnumeric/mlinux-4.0.1/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/lorawan-server/0.6.5-r0/git/src/lorawan_connector_mqtt.erl"},{line,160}]},{gen_server,try_dispatch,4,[{file,"gen_server.erl"},{line,637}]},{gen_server,handle_msg,6,[{file,"gen_server.erl"},{line,711}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,249}]}]}
2019-02-15 17:44:25 =CRASH REPORT====
  crasher:
    initial call: lorawan_connector_mqtt:init/1
    pid: <0.352.0>
    registered_name: []
    exception error: {function_clause,[{lorawan_connector,node_to_vars,[{node,<<2,11,39,119>>,<<"gabriel_profile_new">>,<<"Bleu">>,<<64,82,246,107,80,66,197,145,120,167,91,56,63,94,36,130>>,<<84,137,32,197,119,148,142,187,183,127,243,140,228,137,62,181>>,<<"Mike">>,undefined,3,4,{{2019,2,15},{14,51,54}},{{2019,2,15},{14,51,54}},0,{{2019,2,15},{14,52,45}},[{<<0,128,0,0,160,0,18,168>>,{rxq,867.7,<<"SF12BW125">>,<<"4/5">>,{{2019,2,15},{14,51,54.402131}},1234277533401,2641425236,-45,7}}],1,{undefined,undefined,undefined},{0,0,[{0,7}]},[],undefined,{0,0,869.525},[],[{-45,7},{-67,6.8},{-44,5.8},{-42,6.5}],undefined,undefined,undefined,[],undefined,undefined,undefined,undefined}],[{file,"/home/transnumeric/mlinux-4.0.1/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/lorawan-server/0.6.5-r0/git/src/lorawan_connector.erl"},{line,15}]},{lorawan_connector_mqtt,'-build_hierarchy/3-fun-0-',4,[{file,"/home/transnumeric/mlinux-4.0.1/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/lorawan-server/0.6.5-r0/git/src/lorawan_connector_mqtt.erl"},{line,52}]},{lists,foldl,3,[{file,"lists.erl"},{line,1263}]},{lorawan_connector_mqtt,handle_info,2,[{file,"/home/transnumeric/mlinux-4.0.1/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/lorawan-server/0.6.5-r0/git/src/lorawan_connector_mqtt.erl"},{line,160}]},{gen_server,try_dispatch,4,[{file,"gen_server.erl"},{line,637}]},{gen_server,handle_msg,6,[{file,"gen_server.erl"},{line,711}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,249}]}]}
    ancestors: [lorawan_connector_sup,<0.346.0>,<0.335.0>,<0.303.0>]
    message_queue_len: 0
    messages: []
    links: [<0.347.0>,<0.97.0>]
    dictionary: []
    trap_exit: true
    status: running
    heap_size: 6772
    stack_size: 27
    reductions: 36511
  neighbours:
2019-02-15 17:44:25 =SUPERVISOR REPORT====
     Supervisor: {local,lorawan_connector_sup}
     Context:    child_terminated
     Reason:     {function_clause,[{lorawan_connector,node_to_vars,[{node,<<2,11,39,119>>,<<"gabriel_profile_new">>,<<"Bleu">>,<<64,82,246,107,80,66,197,145,120,167,91,56,63,94,36,130>>,<<84,137,32,197,119,148,142,187,183,127,243,140,228,137,62,181>>,<<"Mike">>,undefined,3,4,{{2019,2,15},{14,51,54}},{{2019,2,15},{14,51,54}},0,{{2019,2,15},{14,52,45}},[{<<0,128,0,0,160,0,18,168>>,{rxq,867.7,<<"SF12BW125">>,<<"4/5">>,{{2019,2,15},{14,51,54.402131}},1234277533401,2641425236,-45,7}}],1,{undefined,undefined,undefined},{0,0,[{0,7}]},[],undefined,{0,0,869.525},[],[{-45,7},{-67,6.8},{-44,5.8},{-42,6.5}],undefined,undefined,undefined,[],undefined,undefined,undefined,undefined}],[{file,"/home/transnumeric/mlinux-4.0.1/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/lorawan-server/0.6.5-r0/git/src/lorawan_connector.erl"},{line,15}]},{lorawan_connector_mqtt,'-build_hierarchy/3-fun-0-',4,[{file,"/home/transnumeric/mlinux-4.0.1/build/tmp/work/arm9 [... cut ...]
     Offender:   [{pid,<0.418.0>},{id,connectors},{mfargs,{lorawan_connector_sup,start_link,[]}},{restart_type,permanent},{shutdown,infinity},{child_type,supervisor}]

2019-02-15 17:44:38 =SUPERVISOR REPORT====
     Supervisor: {<0.335.0>,lorawan_sup}
     Context:    child_terminated
     Reason:     shutdown
     Offender:   [{pid,<0.400.0>},{id,backends},{mfargs,{lorawan_backend_sup,start_link,[]}},{restart_type,permanent},{shutdown,infinity},{child_type,supervisor}]

2019-02-15 17:44:38 =SUPERVISOR REPORT====
     Supervisor: {<0.335.0>,lorawan_sup}
     Context:    shutdown
     Reason:     reached_max_restart_intensity

````

gotthardp, 2019-02-15:
Clear, thanks for the report. It is a bug, I will fix that tonight.

gotthardp, 2019-02-15:
I made a fix. The 0.6.6 is **not** released, however I updated ` lorawan-server_0.6.6.bb` in `meta-lorawan` to include this fix. Once you confirm the bug is fixed (and no other appeared) I will release that as the 0.6.6.

hallard, 2019-02-18:
@gotthardp the 0.6.6 version has the bug fixed, changing connector on web interface does not crash anymore.
However, I don't receive any data with MQTT. I tried on 2 different MQTT servers as follow
[image]
When I click on test button I receive the event on each MQTT server so connection is good. But I don't receive any data on MQTT, but receive them with websocket, that's amazing.

I looked into the crash log and looks like an MQTT connector issue, may why I don't receive any data. I created another connector test to another server from scratch, same issue.

```
2019-02-18 11:32:21 =ERROR REPORT====
** Generic server <0.29668.0> terminating
** Last message in was {uplink,{node,<<2,11,39,119>>,<<"gabriel_profile">>,<<"Bleu">>,<<42,10,238,20,162,230,81,30,107,251,49,160,71,105,144,71>>,<<14,158,2,158,11,99,151,50,18,107,17,18,28,214,32,151>>,<<"Mike">>,undefined,36,36,{{2019,2,18},{8,48,45}},{{2019,2,18},{8,48,45}},0,{{2019,2,18},{10,32,21}},[{<<0,128,0,0,160,0,18,168>>,{rxq,868.5,<<"SF12BW125">>,<<"4/5">>,{{2019,2,18},{10,31,34.977032}},1234521113977,659512684,-53,6.0}}],1,undefined,{0,0,[{0,7}]},[],undefined,{0,0,869.525},[],[{-53,6.0},{-71,7.5},{-45,5.2},{-49,6.0},{-91,6.5},{-43,8.0},{-48,6.0},{-54,6.8},{-50,8.8},{-46,8.5},{-47,7.0},{-55,7.5},{-50,7.0},{-46,7.8},{-47,6.2},{-46,6.2},{-75,6.2},{-50,7.2},{-50,6.8},{-51,8.2},{-56,6.0},{-57,7.5},{-48,7.5},{-54,6.2},{-53,6.0},{-46,6.5},{-50,6.0},{-50,6.5},{-51,6.2},{-48,8.8},{-48,6.0},{-54,5.5},{-54,7.5},{-48,6.8},{-21,6.0},{-54,6.2},{-50,6.2}],undefined,undefined,undefined,[],undefined,undefined,undefined,undefined},#{app => <<"gabriel_handler">>,appargs => <<"Bleu">>,appid => <<"gabriel_app_id">>,best_gw => #{lsnr => 6.0,mac => <<0,128,0,0,160,0,18,168>>,rssi => -53,time => {{2019,2,18},{10,31,34.977032}},tmms => 1234521113977},codr => <<"4/5">>,data => <<253,31,92,106,137,133,7,8,96,28,42,61,0,1,0,0>>,datetime => {{2019,2,18},{10,32,21}},datr => <<"SF12BW125">>,devaddr => <<2,11,39,119>>,deveui => <<0,128,0,0,4,0,31,84>>,fcnt => 36,freq => 868.5,lsnr => 6.0,port => 1,rssi => -53}}
** When Server state == {state,{connector,<<"gabriel-test">>,<<"gabriel_handler">>,<<"json">>,<<"mqtt://192.168.1.8:1883">>,undefined,<<"/gabriel/up">>,<<"/gabriel/events">>,undefined,undefined,undefined,true,undefined,undefined,undefined,undefined,undefined,undefined,undefined,[],0,0,undefined},[{<<"mqtt://192.168.1.8:1883">>,[]},undefined,undefined,undefined],undefined,{<<"/gabriel/up">>,[]},{<<"/gabriel/events">>,[]},undefined,[{[<<"mqtt://192.168.1.8:1883">>,undefined,undefined,undefined],<0.29669.0>,[],{costa,connected311,[{host,"192.168.1.8"},{port,1883},{logger,warning},{keepalive,0},{client_id,undefined},{username,undefined},{password,undefined}],{{2019,2,18},{10,30,17}},0}}]}
** Reason for termination ==
** {function_clause,[{lorawan_connector,node_to_vars,[{node,<<2,11,39,119>>,<<"gabriel_profile">>,<<"Bleu">>,<<42,10,238,20,162,230,81,30,107,251,49,160,71,105,144,71>>,<<14,158,2,158,11,99,151,50,18,107,17,18,28,214,32,151>>,<<"Mike">>,undefined,36,36,{{2019,2,18},{8,48,45}},{{2019,2,18},{8,48,45}},0,{{2019,2,18},{10,32,21}},[{<<0,128,0,0,160,0,18,168>>,{rxq,868.5,<<"SF12BW125">>,<<"4/5">>,{{2019,2,18},{10,31,34.977032}},1234521113977,659512684,-53,6.0}}],1,undefined,{0,0,[{0,7}]},[],undefined,{0,0,869.525},[],[{-53,6.0},{-71,7.5},{-45,5.2},{-49,6.0},{-91,6.5},{-43,8.0},{-48,6.0},{-54,6.8},{-50,8.8},{-46,8.5},{-47,7.0},{-55,7.5},{-50,7.0},{-46,7.8},{-47,6.2},{-46,6.2},{-75,6.2},{-50,7.2},{-50,6.8},{-51,8.2},{-56,6.0},{-57,7.5},{-48,7.5},{-54,6.2},{-53,6.0},{-46,6.5},{-50,6.0},{-50,6.5},{-51,6.2},{-48,8.8},{-48,6.0},{-54,5.5},{-54,7.5},{-48,6.8},{-21,6.0},{-54,6.2},{-50,6.2}],undefined,undefined,undefined,[],undefined,undefined,undefined,undefined}],[{file,"/home/transnumeric/mlinux-4.0.1/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/lorawan-server/0.6.6-r0/git/src/lorawan_connector.erl"},{line,15}]},{lorawan_connector_mqtt,connection_for_node,2,[{file,"/home/transnumeric/mlinux-4.0.1/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/lorawan-server/0.6.6-r0/git/src/lorawan_connector_mqtt.erl"},{line,293}]},{lorawan_connector_mqtt,handle_info,2,[{file,"/home/transnumeric/mlinux-4.0.1/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/lorawan-server/0.6.6-r0/git/src/lorawan_connector_mqtt.erl"},{line,194}]},{gen_server,try_dispatch,4,[{file,"gen_server.erl"},{line,637}]},{gen_server,handle_msg,6,[{file,"gen_server.erl"},{line,711}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,249}]}]}
2019-02-18 11:32:22 =CRASH REPORT====
  crasher:
    initial call: lorawan_connector_mqtt:init/1
    pid: <0.29668.0>
    registered_name: []
    exception error: {function_clause,[{lorawan_connector,node_to_vars,[{node,<<2,11,39,119>>,<<"gabriel_profile">>,<<"Bleu">>,<<42,10,238,20,162,230,81,30,107,251,49,160,71,105,144,71>>,<<14,158,2,158,11,99,151,50,18,107,17,18,28,214,32,151>>,<<"Mike">>,undefined,36,36,{{2019,2,18},{8,48,45}},{{2019,2,18},{8,48,45}},0,{{2019,2,18},{10,32,21}},[{<<0,128,0,0,160,0,18,168>>,{rxq,868.5,<<"SF12BW125">>,<<"4/5">>,{{2019,2,18},{10,31,34.977032}},1234521113977,659512684,-53,6.0}}],1,undefined,{0,0,[{0,7}]},[],undefined,{0,0,869.525},[],[{-53,6.0},{-71,7.5},{-45,5.2},{-49,6.0},{-91,6.5},{-43,8.0},{-48,6.0},{-54,6.8},{-50,8.8},{-46,8.5},{-47,7.0},{-55,7.5},{-50,7.0},{-46,7.8},{-47,6.2},{-46,6.2},{-75,6.2},{-50,7.2},{-50,6.8},{-51,8.2},{-56,6.0},{-57,7.5},{-48,7.5},{-54,6.2},{-53,6.0},{-46,6.5},{-50,6.0},{-50,6.5},{-51,6.2},{-48,8.8},{-48,6.0},{ [... cut ...]
    links: [<0.29371.0>]
    dictionary: [{random_seed,{9829,3379,10169}}]
    trap_exit: true
    status: running
    heap_size: 2586
    stack_size: 27
    reductions: 61424
  neighbours:
    neighbour: [{pid,<0.29371.0>},{registered_name,[]},{initial_call,{emqttc_socket,receiver,2}},{current_function,{erlang,hibernate,3}},{ancestors,[]},{message_queue_len,0},{links,[<0.29369.0>,#Port<0.2386>]},{trap_exit,false},{status,waiting},{heap_size,6},{stack_size,0},{reductions,1673},{current_stacktrace,[]}]

```

gotthardp, 2019-02-18:
Alright. Another bug, another fix was made. I should have done more testing.
I did update the 0.6.6 release in `meta-lorawan`. Coul you rebuild it, please? (The version number is unchanged, it only points to newer sources.)

hallard, 2019-02-18:
petr,
looks fixed, yeahh thanks. 
Anyway not incrementing version number from `meta-lorawan` is confusing bitbake because even with
```
bitbake -c cleanall -f mlinux-lorawan-image
bitbake mlinux-lorawan-image
```
bitbake looks recompiling the lorawan-server (0.6.6) but at the end, installing the .ipk (with force reinstall option) on target didn't work. I suspected having an old version
So I faked and renamed `lorawan-server_0.6.6.bb` to `lorawan-server_9.9.9.bb`

then 
```
bitbake -c cleanall -f mlinux-lorawan-image
bitbake mlinux-lorawan-image
```
and then installed the ipk and all went fine.

I think an option to display the build number (or commit number) in the web interface would really help to debug and be sure we use the correct version.

gotthardp, 2019-02-18:
Sorry for the confusion. I wanted to avoid releasing another buggy version. I will make 0.6.6 from what you have now.

closed 2019-02-18 by gotthardp