gotthardp/lorawan-server #598: WebSocket handle pong

hallard, 2019-02-18:
We can keep websocket connector alive sending 'ping' but looks like the server does not answer 'pong' to ping command so it can be difficult sometimes to handle disconnected. 
Would it be a request feature ?

gotthardp, 2019-02-18:
I guess this is a bug. The server shall respond to ping frames. Are you sure it does not?

hallard, 2019-02-18:
I tried in my code, no success, so tried with chrome websocket extension
It receive server data but does not seems to answer to ping

[image]

hallard, 2019-02-18:
Works my fault
don't send text string like `ws_client.send('ping');`
but with ping command
`ws_client.ping('ping');`

closed 2019-02-18 by hallard