gotthardp/lorawan-server #600: Unable to get data using REST API with Basic Auth and AJAX from browser

hallard, 2019-02-19:
I'm trying to retrieve informations from REST API from a web page. After fighting several hours to find that the request is blocked by CORS security policy on browser side I added all the needed stuff to bypass this. but still no luck.

So I added the CORS "magic" in the LoRaWAN server config as extra headers
sys.config
```erl
    % headers to be added to each HTTP response
    {http_extra_headers, #{
        <<"Access-Control-Allow-Origin">> => <<"*">>,
        <<"Access-Control-Allow-Methods">> => <<"POST, GET, PUT, DELETE, OPTIONS">>,
        <<"Access-Control-Allow-Headers">> => <<"Authorization, Content-Type, X-Custom-Header, X-Requested-With, Origin">>,
        <<"Access-Control-Allow-Credentials">> => <<"true">>,
        <<"content-security-policy">> => <<"default-src * data: 'unsafe-inline' 'unsafe-eval';">>
    }},
```

Then adapted the js code my HTML to allow basic auth
```javascript
	$.ajax({
		type: 'GET',
		url: "http://" + window.location.host + ":8080/api/devices",
		async : false,
		withCredentials: true,
		beforeSend: function (xhr) {
                         xhr.setRequestHeader("Authorization", "Basic " + btoa("admin:admin") );
                },
		success : function(dat) {
			data = dat;
			console.log("GetJSON OK");
		},
		error: function (xhr,ajaxOptions,throwError){
		  console.log(throwError);
  		console.log("GetJSON KO");
		},
	});
```

Since nothing was working agin, I disabled the CORS security policy in browser but still no luck, looks like browser ask server for OPTIONS preflight available and looks not handled by the server itselft, resulting by a 401 unauthorized 

[image]

Did you succeeded to request REST API from a Web page?

Another idea to solve all our issues, would be as discussed in #256 to define a web folder serving custom web pages. Like this request appears coming from same web server and CORS won't impact (tried putting my html files in admin folder and works) and also we don't need to implement another web server on the GW to serve files.

option could be implemented with lorawan config file or on a fixed path, whatever solution is fine for us. 
```erl
    {http_custom_www, [
        {root, "/var/www/html"}
    ]},
```
What do you think?

gotthardp, 2019-02-19:
CORS pre-flight is handled in a wrong way. I have an issue https://github.com/gotthardp/lorawan-server/issues/575 for that, but the implementation is not easy. Adding a custom folder is much easier. Will do that right now.

hallard, 2019-02-19:
Oh, sorry missed that one. CORS is driving me mad from time to time, implemented on browser side, not at bad idea as the beginning but browser should let us add some kind of white list domains !!!

Thanks, will try ASAP, let us know.

gotthardp, 2019-02-19:
Done. There is a `http_custom_web` config option (with an example in the lorawan-server.config). The last argument is a scope that allows you to say who can see that web page.
But you need to build it yourself. Just create a fake bitbake version and set `SRCREV = "d346c9a5ddda60f0d3c954712fb6ad32d34e21bc"`, which is the latest commit.

hallard, 2019-02-19:
Awesome, will try this one

hallard, 2019-02-19:
Is there a priority and a default page in case of conflict, IE if the config is
```
 {"/", <<"/var/www/html">>, [{'*', '*'}]}
```
`http://theserver:8080/index.html` or (`http://theserver:8080/`) will serve /var/www/html/index.html and so to go to admin page I need to `http://theserver:8080/admin/`

gotthardp, 2019-02-19:
There is a priority: the URLs are evaluated top-down (starting with the internal ones) and the first match is served.
For now, it can serve directories only and there is no default file (no switch to index.html). Should there be?

hallard, 2019-02-19:
For now it's not mandatory to have a default html file but could be easier for end user.
I don't know ho to do that, but the idea could be that default http://server:8080 goes to default custom web page pointing by "/" `"/", <<"/var/www/html">>, [{'*', '*'}]}` and only go to lorawan admin on http://server:8080/admin, what's your thought?

gotthardp, 2019-02-19:
What you say should work with `{"/[...]", <<"/var/www/html">>, [{'*', '*'}]}` The `[...]` is a mandatory syntax to accept everything.

hallard, 2019-02-19:
Oh nice to mention thought was to says I can place anything that's why it was not working ;-)
so I confirm it works with config `{"/[...]", <<"/var/www/html">>, [{'*', '*'}]}`

- `http://ip:8080/index.html` fire `/var/www/html/index.html` but ask for authentication ?
- `http://ip:8080/` fire lorawan admin page

Any solution for non admin pages not asking for authentication?

gotthardp, 2019-02-19:
I made another update. Now the config is:
```erlang
[
{"/www", file, <<"/var/www/html/index.html">>, [{<<"anonymous">>, '*'}]},
{"/www/[...]", dir, <<"/var/www/html">>, [{<<"anonymous">>, '*'}]}
]
```
There is a new flag (file or dir). `file` means when user types this URL, serve that file. `dir` means to take what user defined instead of the `[...]` and append it after the dir name.
The scope defines who can do what. The second parameter doesn't make any sense in this context though; it will aways be `'*'`. The first parameter is scope name (see the user administration). It can be `'*'` meaning "any authenticated user", `<<"web-admin">>` meaning "all users with the web-admin role, or any other role, or `<<"anonymous">>` meaning "skip authentication".

gotthardp, 2019-02-19:
One more update: the custom "/" redirection has now higher precedence, so you can define what fires for "/". The admin page fires only when nothing else will be defined.

hallard, 2019-02-19:
Ok thank, done some testing before your last message but was interrupted by phone but
before your last message 
`"/[...]"` broke websocket connection

gotthardp, 2019-02-19:
What was the URL of the websocket that got broken? I will fix that tomorrow.

hallard, 2019-02-19:
In my case it was 
`ws://192.168.1.199:8080/gabriel/up` but I suspect as `/` it was searching in new custom web pages feature

gotthardp, 2019-02-19:
Yeah, the custom web pages are after the admin websites, but before the connectors. The precedence should probably be:
1) internal webpages (admin and api)
2) connectors
3) custom web pages
4) final "/" redirection (http_admin_path)

gotthardp, 2019-02-20:
I fixed the priority, so you should be able to have the custom webpages under "/" without breaking any connectors.

hallard, 2019-02-20:
Awesome, I'll try this ASAP, does it come with a default index.html page or not ?

gotthardp, 2019-02-20:
No default web pages are shipped. Only the admin pages as before.

hallard, 2019-02-21:
@gotthardp 
Just recompiled and tested, all is fine thanks for your help.  Here is my sys.config file
```
    {http_custom_web, [
    % enable this to serve additional web-pages
        {"/", file, <<"/home/mtadm/html/index.html">>, [{<<"anonymous">>, '*'}]},
        {"/[...]", dir, <<"/home/mtadm/html">>, [{<<"anonymous">>, '*'}]}
    ]},
```
Anyway strange things, when trying to serve favicon.ico in the same folder than the other pages, it prompt for authentication.

[image]

may be I should add `/favicon.ico` ?
```
    {http_custom_web, [
    % enable this to serve additional web-pages
        {"/", file, <<"/home/mtadm/html/index.html">>, [{<<"anonymous">>, '*'}]},
        {"/favicon.ico", file, <<"/home/mtadm/html/favicon.ico">>, [{<<"anonymous">>, '*'}]},
        {"/[...]", dir, <<"/home/mtadm/html">>, [{<<"anonymous">>, '*'}]}
    ]},
```

hallard, 2019-02-21:
Tried to add this to sys.config
```
 {"/favicon.ico", file, <<"/home/mtadm/html/favicon.ico">>, [{<<"anonymous">>, '*'}]},
```
same error.

gotthardp, 2019-02-21:
I thought the favicon will be a problem :) I will fix that.

gotthardp, 2019-02-21:
I am sorry than you have to build it yet again. Now you can define your favicon, just like you had above.

hallard, 2019-02-26:
@gotthardp 
That's fine, it works, thanks for fixing

closed 2019-02-26 by hallard