gotthardp/lorawan-server #602: bad_appeui

jot4p, 2019-02-20:
Hi, 

I installed "bosch parking sensors" and registered on the commissioned list of devices:

[image]

I receive:
[image]

What is the problem? 

Thanks

jot4p, 2019-02-21:
If I introduce the AppKey and AppEUI the problem is the same for all devices..

gotthardp, 2019-02-21:
Bad AppEUI is just a check. Did you try to set the AppEUI indicated by the error message?

jot4p, 2019-02-22:
No..  because bosch says that AppEUI is FCD6BD0000190000..

gotthardp, 2019-02-22:
But this is the same number, just with a reverse byte order. Sometimes the hex-strings are reversed.

jot4p, 2019-02-27:
problem solved xD

closed 2019-02-27 by gotthardp

dp512, 2020-05-31:
> problem solved xD

I have exactly same issue "bad_appeui" with Bosch parking sensors. How did u resolve the issue?

altishchenko, 2020-06-01:
> > problem solved xD
> 
> I have exactly same "bad_appeui"-issue with Bosch parking sensors. How did u resolved it?

He reversed the bytes in the AppEUI config parameter :)
FC.D6.BD.00.00.19.00.00 became 00.00.19.00.00.BD.D6.FC
This happens between different node libraries and manufactures.