gotthardp/lorawan-server #603: use a different tile server

pdl86, 2019-02-28:
Hello

would it be possible to use a different tile server, rather than Google maps ? 

idea : enable to enter the url to be queried as a format : i.e

http://IP_address_or_name:port/styles/osm-bright/{z}/{x}/{y}.png

( format of an open tile server  as opentile-gl ).

This can be useful, when the Lorawan server operates in a fully private environment ( no connection to Internet).

Thanks in advance

Regards
Pdl

gotthardp, 2019-02-28:
Hello. Yes, it is possble. The https://github.com/gotthardp/lorawan-server/issues/454 already proposed that.

pdl86, 2019-03-05:
OK , but where in the code can we parameterize it ?

gotthardp, 2019-03-06:
Perhaps I was misleading: I said "it is possible" meaning "it can be done". This feature is however not yet implemented.

gotthardp, 2019-04-04:
I found some time to implement this feature. There is a new config parameter `{map_tile_server, <<"http://{s}.tile.openstreetmap.org/{z}/{x}/{y}.png">>}`.
You can test the latest master, or the future 0.6.8 (to be released in May).

closed 2019-04-04 by gotthardp