gotthardp/lorawan-server #605: Trouble

dencorpz1, 2019-03-04:
Sometimes incoming connections are no longer sent via connections. It helps to restart the service. What could be? Where to see the logs?

gotthardp, 2019-03-06:
The log can be found based on the https://github.com/gotthardp/lorawan-server/blob/master/doc/Troubleshooting.md instructions.

dencorpz1, 2019-03-08:
[crash.log](https://github.com/gotthardp/lorawan-server/files/2946597/crash.log)
[error.log](https://github.com/gotthardp/lorawan-server/files/2946598/error.log)
Сервер встает. порт 8080 на доступ перестает отвечать
что может быть?

gotthardp, 2019-03-08:
The crash looks like https://github.com/gotthardp/lorawan-server/issues/504, which has been fixed in 0.6.2. Are you using some older version?

The error log is full of two types of errors:
 - "Non ASCII string", which means you set "ASCII" handler, but sending non-ASCII characters
 - Topic `<<"gateway/0000e8eb11418c09/rx">>` does not match pattern `<<"gateway/+/tx">>`, which means you have an error in the "Received Topic" field. You have to replace the subscription wildcard (`+`) by a templated value `{somename}` as described in the documentation

closed 2019-03-26 by gotthardp