gotthardp/lorawan-server #606: Purge Database Size for event

hallard, 2019-03-06:
We need to decrease database size, on multitech GW with lot of devices, sometime admin interface does not answer anymore until we delete the following files: 

- `event.DLC` and `event.DCD`
- `rxframe.DLC` and `rxframe.DCD`
- `LATEST.LOG`

As asked in decrease database size #601 We understand we can use parameter 
```
 % amount of rxframes retained for each device/node
    {retained_rxframes, 50},
```
Is there something similar for events ?

Thansks

gotthardp, 2019-03-06:
Yes, there is: `{event_lifetime, 86400}`. The number means how many seconds is an event displayed before it gets deleted.

hallard, 2019-03-27:
Looks like #601 I setup this parameter to 3600 and still have events from 2hours, could it be a localtime or timezone issue?
I've purged both from web interface. Can we purge thru API until it's fixed?

gotthardp, 2019-03-27:
this is really weird, will have a look

the API allows you to delete all entries in a table (that's what the purge buttons do). simply `HTTP DELETE /api/events`, but be careful-- it deletes everything without any warning and any way to recover (that's why its not documented)

autodomain, 2019-03-27:
:(

gotthardp, 2019-03-27:
@hallard To further debug this issue I made the following updates to the latest master:
 * I added new config parameter `{trim_interval, 3600}` to define how often (in seconds) is the database trimmed. Note that if you say {event_lifetime, 3600} and {trim_interval, 3600} your events will never be older than 2 (two!) hours.
 * Upon startup the system (debug) prints `Using config:` following a list of config parameters to make sure nothing was misspelled not misplaced.

hallard, 2019-03-27:
@gotthardp 
Thanks, I'll try this one setting bot to 1800 to see, but not today, we need to finalise production by the end of day

hallard, 2019-03-29:
@gotthardp 

Looks like still not working for RX frames
````
    % amount of rxframes retained for each device/node
    {retained_rxframes, 10},
    % events duration on interface
    {event_lifetime, 1800},
    {trim_interval, 1800},

````

Frames pages 458 frames
[image]

Most recent before
[image]

Older before (2 days ago)
[image]

Events looks better despite the fact there are 3 old events but all other are fine
[image]

gotthardp, 2019-03-29:
Concerning the events: please see the *Last Occurred* date. There are no old events.

gotthardp, 2019-03-29:
Concerning the Frames: what you posted is not necessarily wrong, could you please (wait 1800 seconds after server start and then) filter out only one DevAddr? Is there a DevAddr that has more than a high number of old frames or not?

jot4p, 2020-07-29:
Hi, 

It's possible to remove all received frames, removing rxframe.DLC and rxframe.DCD ? 
Can I remove this with loraserver running ?? 

What is the best procedure ? Because my rxframes page does not load......

JoobyPM, 2022-12-15:
Dear @gotthardp is it safe to remove rxframe.DLC and rxframe.DCD ?
Or to configure retained_rxframes=0?
Thank you for your reply.