gotthardp/lorawan-server #610: Add GW position on uplink fields handler

hallard, 2019-03-12:
Hi Petr,
Would it be possible when available to add fields for gateway GPS position on best_gw and all_gw fields?
Thanks

Apaisal, 2019-03-14:
I agree! I think It is necessary for geolocation feature.

pdl86, 2019-03-14:
I agree to and in fact that what I was looking for in #391.

gotthardp, 2019-03-15:
Hello. I implemented the two new fields. Will you please test it before I make a release?

hallard, 2019-03-15:
@Petr 
Sure, any chance to add #612 so I can test both on the same time because each time I need to rebuild firmware and flash GW ?

gotthardp, 2019-03-15:
Implementing extra buttons is quite complicated, so that one will not be as easy as this.

pdl86, 2019-03-15:
Hello

working with Charles, as stated testing is not so easy do to time to
rebuild.
Is there a chance that you fix aslo #582 (devices description)  and #599
( Gateway description)

regards
Philippe

closed 2019-03-16 by gotthardp

gotthardp, 2019-03-19:
@hallard, did you also test this? Is everything OK so I can make a release?

hallard, 2019-03-19:
Sorry,
yes it works, sorry for the delay

javobarra, 2020-03-13:
Hi Pietr! Is it possible to make independent mqtt message with Gateway parameters?? Or the only way to send Gateway parameters is when receive a node message?
i created a handler and connector for send via mqtt only Gateway parameters. But i think this is not possible to do it

Thanks