gotthardp/lorawan-server #613: Unable to update device desc field with API

hallard, 2019-03-15:
@gotthardp 

I'm trying to update device desc filed using api and ajax. I can query API without any problem to get device information but as soon as I try to update `desc` field I've got an error

Here the code (authentication is working and setup another place on my html page)

```js
// Set default request parameters for lorawan server 
$.ajax({ type: 'PUT',
		async : true,
		url: "/api/devices/" + deveui,
		dataType: "json",
		contentType: "application/json",
		data : { desc: "testtest" },
		success : function(dat) { 
			console.log(dat); 
		},
});
```

here is the console log

[image]

Any idea ?

gotthardp, 2019-03-15:
And the log in the latter case contains something?

In general, when updating an entry you have to provide ALL fields. Partial updates are not supported.

hallard, 2019-03-16:
Yeah that's what I suspected, unfortunately, I've not all the field on my form and my html table due to some optimization so I can't update all. Never mind, I will remove this feature on my app.

gotthardp, 2019-03-16:
Not so fast :) Update of individual fields can be enabled for selected fields. I believe it is useless for all fields, so I enable it only when needed. So if, you need I can add a few for your app.

hallard, 2019-03-18:
Awesome, here we just need the desc field, but may be the appargs could be an option also.

gotthardp, 2019-03-18:
only for devices or also for nodes and gateways?

gotthardp, 2019-03-18:
The latest HEAD supports a request for an individual field, e.g.
http://localhost/api/devices/3333333666666DDD/appargs
allows you to retrieve or modify "appargs" parameter.

It works for all parameters, but for "nodes" and "devices" only.

hallard, 2019-03-19:
Awesome,
will try it out tomorrow we don't need that for GW.

hallard, 2019-03-19:
Compiled image, tested, works for desc field , not tested for appaprgs;-)

Thanks

closed 2019-03-19 by hallard