gotthardp/lorawan-server #614: Suppress Downlink message from AWS

Tamburasca, 2019-03-22:
Hi Petr,
is there a way to suppress the routing of an AWS message, when we do not wish an update of the LoRaWAN device?

Received Topic (AWS) 
$aws/things/Sandbox/shadow/update/accepted

Excerpt of debug.log
2019-03-22 06:15:42.977 [error] <0.17766.1>@lorawan_connector:match_vars:123 Topic <<36,97,119,115,47,116,104,105,110,103,115,47,83,97,110,100,98,111,120,47,115,104,97,100,111,119,47,117,112,100,97,116,101,47,97,99,99,101,112,116,101,100>> does not match pattern {<<36,97,119,115,47,116,104,105,110,103,115,47,83,97,110,100,98,111,120,47,115,104,97,100,111,119,47,117,112,100,97,116,101,47,97,99,99,101,112,116,101,100>>,[]}
2019-03-22 06:15:42.978 [error] <0.17766.1>@lorawan_application_backend:send_downlink:271 Unknown downlink target: #{data => ok,metadata => #{reported => #{'001B0C98' => #{aci1 => #{timestamp => 1553235342},aci2 => #{timestamp => 1553235342},avi1 => #{timestamp => 1553235342},avi2 => #{timestamp => 1553235342},di1 => #{timestamp => 1553235342},di2 => #{timestamp => 1553235342},di3 => #{timestamp => 1553235342},do1 => #{timestamp => 1553235342},do2 => #{timestamp => 1553235342},ro1 => #{timestamp => 1553235342},ro2 => #{timestamp => 1553235342},timestamp => #{timestamp => 1553235342}}}},state => #{reported => #{'001B0C98' => #{aci1 => 0,aci2 => 0,avi1 => 0,avi2 => 0,di1 => 1,di2 => 1,di3 => 1,do1 => 0,do2 => 0,ro1 => 1,ro2 => 0,timestamp => 1553235342.912}}},timestamp => 1553235342,version => 1082}
2019-03-22 06:15:43.851 [error] <0.335.0>@lorawan_gw_forwarder:handle_info:108 Ignored TX_ACK from 7276FF0039030A17: JSON syntax error:

JSON looks like when subscribed
$aws/things/Sandbox/shadow/update/accepted
Mar 22, 2019 7:15:42 AM +0100
{
  "state": {
    "reported": {
      "001B0C98": {
        "aci1": 0,
        "aci2": 0,
        "avi1": 0,
        "avi2": 0,
        "di1": 1,
        "di2": 1,
        "di3": 1,
        "do1": 0,
        "do2": 0,
        "ro1": 1,
        "ro2": 0,
        "timestamp": 1553235342.912
      }
    }
  },
  "metadata": {
    "reported": {
      "001B0C98": {
        "aci1": {
          "timestamp": 1553235342
        },
        "aci2": {
          "timestamp": 1553235342
        },
        "avi1": {
          "timestamp": 1553235342
        },
        "avi2": {
          "timestamp": 1553235342
        },
        "di1": {
          "timestamp": 1553235342
        },
        "di2": {
          "timestamp": 1553235342
        },
        "di3": {
          "timestamp": 1553235342
        },
        "do1": {
          "timestamp": 1553235342
        },
        "do2": {
          "timestamp": 1553235342
        },
        "ro1": {
          "timestamp": 1553235342
        },
        "ro2": {
          "timestamp": 1553235342
        },
        "timestamp": {
          "timestamp": 1553235342
        }
      }
    }
  },
  "version": 1082,
  "timestamp": 1553235342
}

%Downlink Mapping
fun(Fields) ->
  Temp = maps:get(desired, maps:get( state, Fields ), 'default'),
  if Temp == 'default' -> ok;
  true ->
    Devaddr = lists:nth( 1, maps:keys( Temp ) ),
    Temp1 = maps:get( Devaddr, Temp ),
    Ro1 = maps:get(ro1, Temp1),
    Ro2 = maps:get(ro2, Temp1),
    Fields#{
      devaddr => lorawan_utils:hex_to_binary(atom_to_binary(Devaddr,latin1)),
% <<16#001B0C98:32>>, 
      port => 2,
      data => <<03, Ro1, Ro2>>
    }
  end
end.

Any suggestion?

Cheerio, tamburasca

gotthardp, 2019-03-22:
That will be possible in the coming release.

gotthardp, 2019-03-23:
Now you can build a Downlink mapping function that will in some case return an empty list, which will suppress the downlink. It will be included on 0.6.7, but please test the latest master once you got a chance.

Tamburasca, 2019-03-26:
Hi Petr,

thanks for addressing this issue. I've tested with beta 0.6.7 However, I still get the following two error notes and in case when updating the device a third one is thrown. Does this make sense to you?

"Received Topic" and "Subscribe" are both: $aws/things/Sandbox/shadow/update/accepted

Build Downlink is
fun(Fields) ->
  Temp = maps:get(desired, maps:get( state, Fields ), undefined),
  if Temp == undefined -> [];
  true ->
    Devaddr = lists:nth( 1, maps:keys( Temp ) ),
    Temp1 = maps:get( Devaddr, Temp ),
    Ro1 = maps:get(ro1, Temp1),
    Ro2 = maps:get(ro2, Temp1),
    Fields#{
      devaddr => lorawan_utils:hex_to_binary(atom_to_binary(Devaddr,latin1)),
      port => 2,
      data => <<03, Ro1, Ro2>>
    }
  end
end.

2019-03-26 12:54:46.252 [error] <0.799.0>@lorawan_connector:match_vars:128 Topic <<"$aws/things/Sandbox/shadow/update/accepted">> does not match pattern {{re_pattern,0,0,0,<<69,82,67,80,156,0,0,0,16,0,0,0,1,0,0,0,255,255,255,255,255,255,255,255,0,0,0,0,0,0,0,0,0,0,64,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,131,0,88,27,25,29,97,29,119,29,115,29,47,29,116,29,104,29,105,29,110,29,103,29,115,29,47,29,83,29,97,29,110,29,100,29,98,29,111,29,120,29,47,29,115,29,104,29,97,29,100,29,111,29,119,29,47,29,117,29,112,29,100,29,97,29,116,29,101,29,47,29,97,29,99,29,99,29,101,29,112,29,116,29,101,29,100,25,120,0,88,0>>},[]}

2019-03-26 12:54:46.252 [error] <0.799.0>@lorawan_application_backend:send_downlink:296 Unknown downlink target: #{data => [],metadata => #{reported => #{'001B0C98' => #{aci1 => #{timestamp => 1553604886},aci2 => #{timestamp => 1553604886},avi1 => #{timestamp => 1553604886},avi2 => #{timestamp => 1553604886},di1 => #{timestamp => 1553604886},di2 => #{timestamp => 1553604886},di3 => #{timestamp => 1553604886},do1 => #{timestamp => 1553604886},do2 => #{timestamp => 1553604886},ro1 => #{timestamp => 1553604886},ro2 => #{timestamp => 1553604886},timestamp => #{timestamp => 1553604886}}}},state => #{reported => #{'001B0C98' => #{aci1 => 0,aci2 => 0,avi1 => 0,avi2 => 0,di1 => 1,di2 => 1,di3 => 1,do1 => 0,do2 => 0,ro1 => 1,ro2 => 1,timestamp => 1553604886.205}}},timestamp => 1553604886,version => 2318}

2019-03-26 13:04:47.007 [error] <0.344.0>@lorawan_gw_forwarder:handle_info:108 Ignored TX_ACK from 7276FF0039030A17: JSON syntax error:

gotthardp, 2019-03-26:
This may be multiple errors in one.
1) Please use `#` as Received Topic (or upgrade to latest Master). This should solve the "does not match" problem.
2) Are you sure for Fun works correctly? Could you try a nicer version of it
```erlang
fun(#{ state := #{ desired := Temp} } = Fields) ->
  Devaddr = lists:nth( 1, maps:keys( Temp ) ),
  Temp1 = maps:get( Devaddr, Temp ),
  Ro1 = maps:get(ro1, Temp1),
  Ro2 = maps:get(ro2, Temp1),
  Fields#{
    devaddr => lorawan_utils:hex_to_binary(atom_to_binary(Devaddr,latin1)),
    port => 2,
    data => <<03, Ro1, Ro2>>
  };
(_Else) ->
  []
end.
```

Tamburasca, 2019-03-26:
Hi Petr,

I've upgraded to your latest master, updated the fun as you suggested above (sorry I know Python, not Erlang) and get still 

2019-03-26 15:32:43.972 [error] <0.1253.0>@lorawan_application_backend:send_downlink:296 Unknown downlink target: #{data => [],metadata => #{reported => #{'001B0C98' => #{aci1 => #{timestamp => 1553614363},aci2 => #{timestamp => 1553614363},avi1 => #{timestamp => 1553614363},avi2 => #{timestamp => 1553614363},di1 => #{timestamp => 1553614363},di2 => #{timestamp => 1553614363},di3 => #{timestamp => 1553614363},do1 => #{timestamp => 1553614363},do2 => #{timestamp => 1553614363},ro1 => #{timestamp => 1553614363},ro2 => #{timestamp => 1553614363},timestamp => #{timestamp => 1553614363}}}},state => #{reported => #{'001B0C98' => #{aci1 => 0,aci2 => 0,avi1 => 0,avi2 => 0,di1 => 1,di2 => 1,di3 => 1,do1 => 0,do2 => 0,ro1 => 0,ro2 => 0,timestamp => 1553614363.839}}},timestamp => 1553614363,version => 2460}

gotthardp, 2019-03-26:
OMG, I had a bug there. Please try again now. Sorry for that.

Tamburasca, 2019-03-27:
Petr,

no problem at all. I've upgraded and the errors are gone. We'll keep testing. The sole error remaining is probably related to our Kerlink-Gateway. We'll need to investigate.
2019-03-27 08:51:29.995 [error] <0.354.0>@lorawan_gw_forwarder:handle_info:108 Ignored TX_ACK from 7276FF0039030A17: JSON syntax error:

Let me seize the opportunity to appreciate the fine work of yours. Cheerio, tamburasca

gotthardp, 2019-03-27:
Would you be able to figure out what characters is the gateway sending in the TX_ACK?

This packet may include an optional JSON. Your gateway did send some whitespace characters, but no JSON. I can silence this too, but it would be better to undestand the gateway behaviour. I have a Kerlink too and it is behaving well.
https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT#L336

gotthardp, 2019-03-27:
It's a quite simple change, so I did it anyway. The error should disappear now.

xypron, 2019-03-29:
string:trim(<<16#00>>) results in <<0>>. Please, provide a solution that also eliminates <<0>> as sent by some gateways, e.g. Kerlink Wirnet iFemtoCell.

gotthardp, 2019-04-03:
I believe v0.6.7 solved it all.

closed 2019-04-03 by gotthardp