gotthardp/lorawan-server #615: Question about ADR support

elisascheer, 2019-03-22:
Hi,

I am currently testing LoRa network with Adeunis field test device. The device is configured with ADR support but it always transmits frames with SF12  and doesn't adapt to network condition even if ADR is ON on the server admin.

Do you have any idea?

Thanks

altishchenko, 2019-03-22:
@elisascheer There are a few things to be tweaked:
1. There is a configuration parameter frames_before_adr which you can set to a low value (i am using 10), like this {frames_before_adr, 10} in the server sys.config file (or in lorawan_server.config if you are building from source).
2. On the Adenius device Profile page make sure the ADR is set to Auto-Adjust, Set Datarate parameter is set to SF7BW125, you can also set power here.

elisascheer, 2019-03-22:
Ok so I added the line {frames_before_adr, 10} in sys.config so that my file looks like this : 

`[
% server configuration
{lorawan_server, [
    % update this list to add/remove applications
    {plugins, [
        {<<"semtech-mote">>, lorawan_application_semtech_mote},
        {<<"microchip-mote">>, lorawan_application_microchip_mote},
        {<<"backend">>, lorawan_application_backend},
        {<<"websocket">>, lorawan_application_websocket}]},
    % UDP port listening for packets from the packet_forwarder Gateway
    {packet_forwarder_listen, [{port, 1680}]},
    {frames_before_adr, 10},
    % HTTP port for web-administration and REST API
    {http_admin_listen, [{port, 8080}]},
    % HTTP/SSL port
    % {http_admin_listen_ssl, [
    %     {port, 8443},
    %     {certfile, "cert.pem"},
    %     {keyfile, "key.pem"}
    % ]},
    % default username and password for the admin interface
    {http_admin_credentials, {<<USER_VAR>>, <<PASS_VAR>>}},
    % amount of rxframes retained for each device/node
    {retained_rxframes, 50},
    % websocket expiration if client sends no data
    {websocket_timeout, 3600000} % ms
]},
% error logging settings
{lager, [
    {colored, true},
    {handlers, [
        % {lager_console_backend, debug},
        {lager_file_backend, [{file, "log/debug.log"}, {level, debug}]},
        {lager_file_backend, [{file, "log/error.log"}, {level, error}]}
    ]}
]},
{sasl, [
    {errlog_type, error}
]}].
`
On the server admin interface, I set ADR to ON just like this :  

[image]

And the device is still sending with SF12. The device has been configured to use ADR (I checked on IoT configurator app).
Maybe I forgot something but I don't know what

elisascheer, 2019-03-22:
update : it's working now thanks for your help !

altishchenko, 2019-03-22:
@elisascheer The server needs at least 10 frames, by your config, to start ruling the ADR :) What version of the server you are running? ADR settings tabs were changed for 0.6.X line of code.

elisascheer, 2019-03-25:
Oh okay the version which is installed is the 0.4.7

gotthardp, 2019-03-25:
The `frames_before_adr` option is supported since v0.5.3. In your version you need to wait 20 frames after the last ADR change before the auto-ADR becomes effective.

elisascheer, 2019-03-26:
I noticed that when I was testing with the network tester. 
What I find strange is that the first time I tested the auto-ADR, it worked but later when the network server recalculates the averages of rssi and snr it changes data rate but it didn't change on the end node (still sends in SF 12)
[image]

gotthardp, 2019-03-26:
In the debug.log there is a reason why the ADR did what it did. Please provide it to us so we can check.

elisascheer, 2019-03-26:
Output of debug.log file :

[image]

elisascheer, 2019-03-26:
tested device is 60047746

elisascheer, 2019-03-26:
I understand what's happening, there is no `send_adr` in the debug.log file. So I found these lines when it was working :

[image]

But I dont't know why it doesn't send these messages anymore

elisascheer, 2019-03-26:
As you suggested I upgraded to the latest version 0.6.6 so I added the field `{frames_before_adr,10}` in the `sys.config` and I can see in the debug.log file :

`2018-03-22 11:46:04.283 [debug] <0.855.0>@lorawan_mac_commands:calculate_adr:336 DataRate 60047746: average snr 8 -10.0 = 18, dr 0 -> step 6
2018-03-22 11:46:04.283 [debug] <0.855.0>@lorawan_mac_commands:calculate_adr:346 Power 60047746: average rssi -33, power 1 -> up by 16
2018-03-22 11:46:04.284 [debug] <0.855.0>@lorawan_mac_commands:send_adr:370 LinkADRReq {10,6,[{0,2}]}
2018-03-22 11:46:04.284 [debug] <0.855.0>@lorawan_mac_commands:build_fopts:77 60047746 <- [{link_adr_req,6,10,7,0,0}]
`
And after there are these messages : 

`2018-03-22 11:51:34.429 [debug] <0.1265.0>@lorawan_mac_commands:handle_fopts:16 60047746 -> [{link_adr_ans,0,0,1}]
2018-03-22 11:51:34.429 [warning] <0.1265.0> node 60047746 {adr_req_failed,{0,0,1}}
2018-03-22 11:51:34.429 [warning] <0.1265.0>@lorawan_db_guard:send_alert:196 node 60047746 [{<<"linkadr_failed">>,25}] []`

gotthardp, 2019-03-26:
Cool! This `linkadr_failed` means your device doesn't like the Power and DataRate the server sent.

elisascheer, 2019-03-26:
Thank you!
It's strange because it worked once so you think it's the device configuration and not the configuration of the network server (such as the network section) ?

gotthardp, 2019-03-26:
It is refused by the device, so ideally if you could debug the firmware and figure out why it didn't like it. Of course, the device could refuse it because the Network, Provide or Device config is wrong, but it's the device that didn't like it.

elisascheer, 2019-03-27:
Ok it's working now, my device can send with only 14 dBm and so can't send with 2 dBm as suggested the network server, I changed my network configuration and it's now sending with adapted datarate, I close this issue.
Thanks for your help!

closed 2019-03-27 by elisascheer