gotthardp/lorawan-server #616: NetID SubID clarification

hallard, 2019-03-23:
Hi @gotthardp 

I always wanted to ask but never had concrete example for. But now I have one and I'm not sure to understand linking between NetID and SubID.

Documentation state 
```
The Network NetID and Group SubID are used to create DevAddr of OTAA devices.
Each DevAddr is composed of 7 LSB of NetID, followed by X SubID bits, 
followed by 25-X random bits. 
This allows operator to define separate private sub-networks using the same NetID.
```
default value example in field is `0:3` and I have set a SubID network with `12:8` (not sure where does it come from) but I don't understand who it works.

Here my use case, I got a private network with NetID set to 00001
Now I need to have devaddr with last byte to be unique (don't care the values before)
aabbccXX. I understand that on this network I will be limited to 255 devices and this is exactly what I need. So what the values I need to put in SubID to have for example (I give you 3 examples to be sure to understand):

- devaddr going throuh `aabbcc00` to `aabbccFF`
- devaddr going throuh `aabbcc00` to `aabbcc7F`
- devaddr going throuh `aabbcc80` to `aabbccFF`

Once again, I really don't care of value aabbcc, my focus is that I want to be sure that last byte of devaddr for my SubID is different and going from 0x00 to 0xFF (example 1) or from 0x00 to 0x7F (example 2) or from 0x80 to 0xFF (example 3)

Thanks for any clarification

gotthardp, 2019-03-23:
The DevAddr is 32 bits, 7+X fixed, followed by 25-X random. To have the last 8 bits random (0x00-0xFF) X=17, for having last 7 bits random (0x7F-0xFF) X=18. 0x00-0x80 is not possible because that would require some fixed bits at the end. The random are always the last.

The SubID is in the form of Z:X, where Z is the fixed hexa value and X is the number of bits in it. (Thats the length of extra fixed part explainef above.) For example, 0:3 is 3 zero bits, 0:17 is 17 zero bits.

Does it make sense?

hallard, 2019-03-23:
Yes, it makes me better understanding, so with a NetID 0001F (7 LSB to 1)
and SubID set to `1234500:17` devaddr will be `FF234500` to `FF2345FF`
and with SubID set to `1234500:18` devaddr will be `FF23457F` to `FF2345FF` is that correct ?

gotthardp, 2019-03-25:
The bits are concatenated, not *and*-ed, hence you don't need to add the two zeros at the end.

If NetID=0x0001F, then the last 7-bits are "0011111".
If SubID=0x12345, then its last 17-bits are "‭10010001101000101‬".
If we take the 7bits, followed by the 17bits we get "0011111‭10010001101000101‬", which is ‭0x3F2345‬.
The remaining 8bits (32-7-17) will be random, hence the devaddr will be `3F234500` to `3F2345FF`.

gotthardp, 2019-03-25:
Likewise,
If NetID=0x0001F, then the last 7-bits are "0011111".
If SubID=0x12345, then its last 18-bits are "‭010010001101000101‬".
If we take the 7bits, followed by the 18bits we get "0011111‭010010001101000101‬"‬, i.e. 25-bits (its conversion to hexa could be confusing).
The remaining 7bits (32-7-18) will be random, hence the devaddr will be `‭3E91A280‬` to `3E91A2FF`.

hallard, 2019-03-25:
@gotthardp 
Makes senses the trick was "concatenated" and and'ed ;-)

So my example 2 

If NetID=0x0001F, then the last 7-bits are "0011111".
If SubID=0x12346, then its last 18-bits are "‭010010001101000110‬".
If we take the 7bits, followed by the 18bits we get "0011111‭010010001101000110‬"‬, i.e. 25-bits (its conversion to hexa could be confusing).
The remaining 7bits (32-7-18) will be random, hence the devaddr will be ‭3E91A200‬ to 3E91A27F.

I admit that taking 8bits of NetID instead of 7 would have made binary calculation easier ;-)

Thanks for clarification

gotthardp, 2019-03-25:
The "7-bits of NetID" is per LoRaWAN standard. That cannot be changed.

closed 2019-03-25 by gotthardp