gotthardp/lorawan-server #618: API - Create new commissioned device

jot4p, 2019-03-27:
Hi, 

[image]

What is the complete command to put information via API and create a new commissioned device ?

gotthardp, 2019-03-27:
Well, do `HTTP GET /api/devices/xzy` and you will see a JSON structure that is corresponding to the web admin fields. Create a JSON with the mandatory fields and do `HTTP POST /api/devices`.

jot4p, 2019-03-27:
{
"appeui":"XXXXXXXXXXXXXX",
"appkey":"XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX",
"deveui":"XXXXXXXXXXXXXXXXX",
"node":"E0E0E0E0",
"profile":"device_profile"
} 

ok, like that.

gotthardp, 2019-03-27:
OK like this
Even the "node" is not needed. If not provided it is generated upon join.

closed 2019-03-28 by gotthardp