gotthardp/lorawan-server #619: API request thru MQTT ?

hallard, 2019-03-27:
@gotthardp 

May be a stupid question, but could all the API stuff that we can do thru HTTP could also be exposed thru MQTT?
Could be interesting requesting devices directly thru MQTT like
/api/devices/123`

And would avoid lot of AJAX Request from browser and managing authentication on each.

What's your thought?

gotthardp, 2019-03-27:
#550 requested an API via WebSocket, would this be a suitable alternative?

hallard, 2019-03-27:
Today I'm using it over HTTP so it's not really an issue but I won't use it over websocket. BTW what about authentication over WS?

MQTT would be a feature, don't worry about it it's not on my "current need" but I think it's cleaner than WS (from my side). Just set host, auth user pass and that's it, do able from many app.