gotthardp/lorawan-server #620: Export button and Uplink are not working.

BouTayehGaby, 2019-03-28:
The uplink was working fine when two sensors were sending simultaneously, I was able to send data to my application server (Web application on a hosting server).

I deployed 15 sensors and the Network server stopped transmitting the received frames to my application server.

Therefore I tried to use the export button in Frames but it is not working!

current version : 0.6.3.

Thank you

gotthardp, 2019-03-28:
Hello. I confirm the Export of rxframes is not working. I just fixed that.
To debug the "uplink not working" issue I'd need more information like debug logs etc.

BouTayehGaby, 2019-03-29:
[image]

As seen in the image, on the recieving server side (where my application is located) my IP was black listed since the requests header contained illegal characters (Characterized as an intrusion attack on the server).  
It is weird since I was able to send data previously without any problem.

Any ideas what could have generated such an issue?

gotthardp, 2019-03-29:
I cannot tell you more than what you can read in the log you posted. :-) I will close this issue now as the first problem was fixed and the second is not related to the lorawan server.

BouTayehGaby, 2019-03-29:
isn't it however related to the content of the header in the transmitted packet generated by the lorawan server?

gotthardp, 2019-03-29:
true, but do you know what did the server send?

BouTayehGaby, 2019-03-29:
Which server are you referring to? the lorawan server?

gotthardp, 2019-03-29:
yes. Could you please track what the lorawan-server did send that made the IDS to stop the traffic?